RobotLocomotion/drake #9811: Resurrect double_pendulum example using MultibodyPlant

sherm1, 2018-10-25:
PR #9800 removes dead parsing code. That requires removing examples/double_pendulum which is the only example that was using that dead code. That is actually one of the nicer "hello world" Drake examples, with a beautiful README and short, readable implementation.

Luke Peterson (@hazelnusse) volunteered to try reworking the example to use MultibodyPlant while retaining its character. I'll mark this as high priority since we don't want to lose the example, though we do want to nuke the dead code.

sherm1, 2018-10-25:
Luke, after #9800 merges the double_pendulum example will still be functional, but using outdated custom parsing and soon-to-be-deprecated RigidBodyPlant rather than the current MultibodyPlant. Your mission, should you choose to accept it, would be to modernize this and delete the custom parsing files.

sherm1, 2018-12-11:
@hazelnusse -- Luke, are you going to be able to get to this or would it be better if we reassign it?

sherm1, 2018-12-12:
@mitiguy, please resurrect this example in modernized form when you have a chance.

sherm1, 2019-02-07:
Is this issue still active, @mitiguy ?

jwnimmer-tri, 2020-02-11:
I'm going to remove the example for now -- git remembers, so we can always resurrect a rewrite later.  I'll let you decide whether to close the issue or not.

jwnimmer-tri, 2021-11-12:
This code no longer exists on the main branch.

closed 2021-11-12 by jwnimmer-tri