RobotLocomotion/drake #9817: Does the Schunk WSG use an integral term (PID) in its internal controller

RussTedrake, 2018-10-26:
We should know the answer (and model it properly).

The test should be simple enough: Put an object inside the open gripper and close the gripper by sending a position command that is slightly smaller than the actual size of the object.  Then decrease the commanded position towards zero.  

If there is an integral term, then I would expect that at steady-state the gripper will output force ==force_limit for *all* positions < than the object width.  If it is only PD control, then I would expect that the deeper the commanded penetration, the larger the output force until it eventually hits the force_limit.

sammy-tri, 2018-10-26:
Based on my testing (some done quite a while ago while clamping down on a scale, and general observations), the commanded force (which, since we don't use the smart fingers, probably means something more like "desired motor current") is the primary thing being controlled, with the commanded position having a role more like "stop applying current to the motor if you actually get here and just maintain a position).

To answer your question above, yes, output force == force limit for all positions less than the object with, but I don't have a good sense if that implies an integral term based on finger position or if it implies a different control model in which force (motor current) is the property being controlled.

calderpg-tri, 2018-10-26:
For additional reference, on page 20 or the [WSG manual](https://schunk.com/fileadmin/pim/docs/IM0004935.PDF) it shows the coupling between gripper speed and gripper force, as well as the effects of grasp location.

If you don't have force-sensing fingers attached, the "measured force" is an estimate based on motor current, but this is effected by the finger speed.

sammy-tri, 2018-10-26:
I retested by sending `lcmt_schunk_wsg_command` messages directly to the driver while a rigid object was positioned between the gripper fingers.  The status messages from the gripper reported that the gripper was applying the same commanded force for any commanded position where the object was clamped between the fingers.  (the object was approximately 80mm wide, and values from slighty less than 80mm down to 10 all indicated the commanded force).

RussTedrake, 2018-10-28:
Thanks all.  A few more thoughts after looking at the manual and talking to Sam.
1 - the WSG firmware is a bit "smarter" than I would like (to model).  It transitions between discrete modes for moving, grasping, etc.  
2 - sam said we do send a velocity command to the robot, but always send max velocity... that grant's initial experiments with the hardware suggested that we would never want anything else.

the current model / plan of record, is that a very stiff PD controller w/ saturation is an approximation of the above.  but i will look a bit more at the driver implementation before I consider this issue completely resolved for myself.

RussTedrake, 2018-12-26:
per conversation with @siyuanfeng-tri (who was working on this last week), we think that perhaps something like a bang-bang controller (with velocity and acceleration saturation) might be the right solution for the *controller* (replacing a smoother that sends commands to a PD/PID controller).

See #10274