RobotLocomotion/drake #9827: drake visualizer has normals flipped for box primitives

RussTedrake, 2018-10-27:
our current manipulation_station example uses a lot of box primitives for visualizing the station, etc.  and I've been noticing that the normals on the box primitive, as rendered through the drake_visualizer interface are consistently bad.  so far, it's only made for aesthetic problems, but now I'm trying to manipulate boxes and the objects often dissappear into the table / surrounding structure.

This is the station viewed from a nominal (human-relevant) viewing angle.  It's supposed to be a table with 80-2 vertical trusses.  But the closest leg is disappearing into the base (+ other artifacts)

[image]

this is the same station, viewed from a slightly lower camera angle (but still above the table/ground).  now you can see that there was actually a "foam brick" on the table the entire time.  it was just being hidden by the visualization artifacts.

[image]

you can also see into the sides of the table surface, which should be a solid box.
is there a reason that this might have changed?  vtk version bump?

EricCousineau-TRI, 2018-10-27:
Running `//examples/manipulation_station:proof_of_life` on my machine on current master (81352d8) did not indicate odd normals:
> [image]

Perhaps this is specific to how the box primitives are making it in?
Do you have a reproduction model + script that it's easy to run (perhaps a `pydrake` script using `ConnectDrakeVisualzier` + an SDF file)?

RussTedrake, 2018-10-27:
Nice call!  The alpha values were 0.9.  Setting them to 1.0 has resolved it.  You rock.

cc @pangtao22 , @Sproulx -- let's be sure to use alpha = 1.0 from now on.

closed 2018-10-27 by RussTedrake