RobotLocomotion/drake #9829: Use gurobi 8.0.1 or 8.1.0

hongkai-dai, 2018-10-27:
Currently, Drake supports gurobi 8.0.0. But on gurobi website, they only provide download for gurobi 8.0.1 and 8.1.0 http://www.gurobi.com/downloads/gurobi-optimizer. We should upgrade our support on gurobi.

jwnimmer-tri, 2018-10-27:
Probably a duplicate of #9624 (or else, that one is a duplicate of this)?

hongkai-dai, 2018-10-27:
Thanks @jwnimmer-tri ! They are related. I think the solution in #9624 is to download gurobi from the link in https://github.com/RobotLocomotion/drake/issues/9624#issuecomment-428044725. On the other hand, we should test `GurobiSolver` with the newer version of Gurobi, possibly 8.1.0. Every upgrade of Gurobi fixes some problem, and likely also improve the performance. I plan to upgrade our GurobiSolver to support 8.1.0 once we finish refactoring MathematicalProgram.

hongkai-dai, 2019-03-18:
According to http://www.gurobi.com/products/whats-new/whats-new-in-the-latest-version, Gurobi 8.1 is significantly faster than Gurobi 8.0 for mixed integer problem. We should hence upgrade to this newer version.

EricCousineau-TRI, 2019-04-25:
Given that Jamie has #11269, reassigning.

EricCousineau-TRI, 2019-05-09:
Per f2f, should be up and coming.

PR #11269 (2019-05-14): Upgrade supported Gurobi version to 8.0.1
Closes #9829.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/11269)
closed 2019-05-14 by jamiesnape