RobotLocomotion/drake #9859: using utime==0 as the check for lcm message arrivals

RussTedrake, 2018-10-30:
@sammy-tri -- when I feature reviewed the lines like:
https://github.com/RobotLocomotion/drake/blob/master/manipulation/schunk_wsg/schunk_wsg_lcm.cc#L44
(and the repeat logic a few lines below) a few days ago, I thought it was ok (thinking that properly initialized utime is always >> 0).  Now I'm questioning that decision.

When sending lcm traffic from sim to sim, we get utime=0 at context.get_time()==0.  Which means the initial_position inputs will take precedence over any actual data published at time 0.  Was that your intention?  I can't immediately think of a better solution, but it feels sub-optimal.

sammy-tri, 2018-10-30:
I was indeed assuming that utime would always be greater than zero for a valid message, and I missed the fact that 0 is a valid time in simulation.  I think the right fix would probably be to initialize utime to -1 in the model value in the constructor and check for that as a sentinel value instead.  I think it probably makes sense to keep this issue open until that's fixed (which I should do soon since it's not that difficult).

jwnimmer-tri, 2018-10-30:
Relying on the model value will probably not help, but in any case would be an incorrect application of how the systems framework is designed to operate.

The LcmSubscriberSystem has a message_count state variable, which you could tie into an output port if that helps.

In the alternative, if the only thing you really want to guard on is whether the lcmt_schunk_wsg_command is value-initialized (all zeros), then just check that -- look at _all_ fields, not just the timestamp.  If they are all zero, then it likely means that the Subscriber has never received a message yet.

sammy-tri, 2018-10-30:
From f2f with Jeremy I think my preferred solution is to create an appropriate `Serializer` with the default command when constructing the `LcmSubscriberSystem` (instead of using the sugar in `LcmSubscriberSystem::Make`) which feeds the command receiver instead of teaching the receiver itself about the initial state.

hongkai-dai, 2019-03-25:
@sammy-tri could you update the status on this issue? Thanks

sammy-tri, 2019-03-25:
The issue is still present.

EricCousineau-TRI, 2020-05-14:
Per Slack convo w/ Sammy, realized that this had tripped me up too ;)
I was fixing an input port with a "fake" Schunk WSG command, and didn't think `utime` would be important. After some debugging (and finding out first that `q0_wsg == [0., 0.]` requires collision filtering), found that this was the reason.