RobotLocomotion/drake #9886: py rbt: `relativeRollPitchYaw` does not return correct values via AutoDiff

EricCousineau-TRI, 2018-10-31:
See the following lines @neeld's gist:
https://gist.github.com/neeld/8f8da067bb019a3ab07da6cf547282d0#file-relative_rpy_err-py-L55-L61

Excerpt:
```
# Print statement results for fli_dbl: 
# [-0.8965 -0.      0.    ]

# Print statement results for fli_ad: 
# [<AutoDiffXd -0.896455004315 nderiv=32>
# <AutoDiffXd -0.896455004315 nderiv=32>
# <AutoDiffXd -0.896455004315 nderiv=32>]
```

Placing at medium priority because it's unlcear if this issue is specific to:
- `RigidBodyTree::relativeRollPitchYaw`
- Python binding of AutoDiff flavor of `relativeRollPitchYaw`
- `RigidBodyTree::relativeTransform`
- `RotationMatrix(Matrix3)`
- `RollPitchYaw(RotationMatrix)`

I will try root cause this and assign the priority after that.

neeld, 2018-11-16:
Hey, just wondering what's happening to this issue?

EricCousineau-TRI, 2018-11-18:
Sorry for the delay! (Had gotten caught up in a few other things.)

I've reproduced the error you saw in #10061 off of latest `master`; I will see if this is an issue with `pybind11`, or with the underlying C++ implementation by reproducing this error in C++.

EricCousineau-TRI, 2018-11-18:
Found the minimal repro, which is that fixed-size matrices are broken in our fork of `pybind11`. Ew...

EricCousineau-TRI, 2018-11-18:
To be fixed in #10061 - thanks again for reporting this, and sorry again about the delay!

neeld, 2018-11-18:
no problem! Thanks for looking into this

PR #10061 (2018-11-20): py autodiff: Ensure fixed-size vectors are converted correctly
Resolves #9886

Requires:
- [x] https://github.com/RobotLocomotion/pybind11/pull/26

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/10061)
closed 2018-11-20 by EricCousineau-TRI