RobotLocomotion/drake #9887: pydrake: Make it easier to figure what index is a TypeSafeIndex vs. GeometryId, add tests

EricCousineau-TRI, 2018-10-31:
From Slack discussion:

> pangtao22 [36 minutes ago]
> How can I get the integer value of `*Index` in python?
> 
> pangtao22 [33 minutes ago]
> Is this the right way to do it: `int(body.model_instance())`?
> 
> jwnimmer-tri [31 minutes ago]
> My guess is yes, from reading the implementation.  I don't see a python unittest showing an example though. (edited)
>
> eric.cousineau [3 minutes ago]
> D'oh, yeah, need to add that in. 
> 
> eric.cousineau [1 minute ago]
> Actually, sorry: Yes, there are tests, but it's a tad opaque and buried in C++: They are type-safe indices, which are tested here:
> https://github.com/RobotLocomotion/drake/blob/27e8ed478f9131d77f625f3ab3a9cd7f90b09fb2/bindings/pydrake/util/test/type_safe_index_pybind_test.cc

\cc @pangtao22 @jwnimmer-tri

SeanCurtis-TRI, 2018-10-31:
It's worth noting that the `Identifier` is designed to be opaque. You're not supposed to know what value it has. (Although you *can* print it out...which leaks that a bit.) But there is no API to directly get the underlying integer value.

pangtao22, 2018-11-01:
`PoseBundle::get_model_instance_id(int)` returns an integer value of the ModelInstanceIndex of a rigid body. The values are used in `MeshcatVisualizer` to differentiate rigid bodies that share the same name but belong to different model instances.

SeanCurtis-TRI, 2018-11-01:
When frames are registered with SceneGraph, they have a `frame_group` which is documented to be mappable to MBP's model instance id. Looking in MBP, it looks like it's `ModelInstanceIndex` is being converted to an int and stored with the registered frame.

However, I need to add a function to introspect the `frame_group`.

EricCousineau-TRI, 2020-05-01:
I don't think there's been any further pain on this specific issue. Closing for now.

Regarding discussion on how to do viz in Meshcat, see this for discussions on how to better improve (I think):
https://github.com/RobotLocomotion/drake/issues/10482#issuecomment-620724731

closed 2020-05-01 by EricCousineau-TRI