RobotLocomotion/drake #9890: py systems: Figure out why `OutputPort.Eval()` with `Value[QueryObject]` causes a segfault

EricCousineau-TRI, 2018-11-01:
Ran into this in #9889:

The following code causes a segfault:
```py
query_object = scene_graph.get_query_output_port().Eval(sg_context)
query_object.ComputePointPairPenetration()
```
but this does not:
```py
query_object = scene_graph.get_query_output_port().EvalAbstract(sg_context).get_value()
query_object.ComputePointPairPenetration()
```

It seems like the `keep_alive` behavior is getting lost somewhere when calling `get_value` via the `pybind` C++ API:
https://github.com/RobotLocomotion/drake/blob/4fba81e9b43eaa465dcd6939172ac854fdccbda9/bindings/pydrake/systems/framework_py_semantics.cc#L235-L239

I'm not sure why lifetime matters here, as the context should exist the same no matter what, and I would assume that `AbstractValue` would be living in the context...

sherm1, 2018-11-01:
Does `get_value()` figure out the right downcast to do based on the type of the return argument?

EricCousineau-TRI, 2018-11-01:
Yuppers; we actually have define an instantiation for each `Value[T]` (done by `AddValueInstantation` in `systems_pybind.h`), and its `get_value()` returns the correct concrete type.

The two code snippets are effectively the same thing; the only difference is that some bits are executed in C++ in the first.

SeanCurtis-TRI, 2018-11-01:
I mentioned this in passing in the PR #9889. But, for `QueryObject` particularly, it may not be a keep alive issue. It may be a copy issue. When copying the `QueryObject`, it becomes invalid. Admittedly, it should throw and not segfault...so, maybe it is a live thing.

commit 12ff141c (2018-11-01): py systems: Fix `OutputPort.Eval()` (Resolves #9890)
closed 2018-11-01 by soonho-tri