RobotLocomotion/drake #9903: Compilation error with python bindings

rueberger, 2018-11-01:
Python bindings fail to compile on the current master with:

```ERROR: /drake/drake/bindings/pydrake/BUILD.bazel:50:1: error executing shell command: 'bazel-out/host/bin/tools/workspace/pybind11/mkdoc $@' failed (Exit 1)
Traceback (most recent call last):
  File "/drake/drake-build/_bazel_/f4edae0fad3231c5cc7e2df04d356805/sandbox/processwrapper-sandbox/1042/execroot/drake/bazel-out/host/bin/tools/workspace/pybind11/mkdoc.runfiles/drake/third_party/com_github_pybind_pybind11/mkdoc.py", line 594, in <module>
    main()
  File "/drake/drake-build/_bazel_/f4edae0fad3231c5cc7e2df04d356805/sandbox/processwrapper-sandbox/1042/execroot/drake/bazel-out/host/bin/tools/workspace/pybind11/mkdoc.runfiles/drake/third_party/com_github_pybind_pybind11/mkdoc.py", line 584, in main
    print_symbols(f, root_name, symbol_tree.root)
  File "/drake/drake-build/_bazel_/f4edae0fad3231c5cc7e2df04d356805/sandbox/processwrapper-sandbox/1042/execroot/drake/bazel-out/host/bin/tools/workspace/pybind11/mkdoc.runfiles/drake/third_party/com_github_pybind_pybind11/mkdoc.py", line 438, in print_symbols
    print_symbols(f, key, child, level=level + 1)
  File "/drake/drake-build/_bazel_/f4edae0fad3231c5cc7e2df04d356805/sandbox/processwrapper-sandbox/1042/execroot/drake/bazel-out/host/bin/tools/workspace/pybind11/mkdoc.runfiles/drake/third_party/com_github_pybind_pybind11/mkdoc.py", line 438, in print_symbols
    print_symbols(f, key, child, level=level + 1)
  File "/drake/drake-build/_bazel_/f4edae0fad3231c5cc7e2df04d356805/sandbox/processwrapper-sandbox/1042/execroot/drake/bazel-out/host/bin/tools/workspace/pybind11/mkdoc.runfiles/drake/third_party/com_github_pybind_pybind11/mkdoc.py", line 438, in print_symbols
    print_symbols(f, key, child, level=level + 1)
  File "/drake/drake-build/_bazel_/f4edae0fad3231c5cc7e2df04d356805/sandbox/processwrapper-sandbox/1042/execroot/drake/bazel-out/host/bin/tools/workspace/pybind11/mkdoc.runfiles/drake/third_party/com_github_pybind_pybind11/mkdoc.py", line 433, in print_symbols
    doc_var, delim, symbol.comment))
  File "/drake/drake-build/_bazel_/f4edae0fad3231c5cc7e2df04d356805/sandbox/processwrapper-sandbox/1042/execroot/drake/bazel-out/host/bin/tools/workspace/pybind11/mkdoc.runfiles/drake/third_party/com_github_pybind_pybind11/mkdoc.py", line 399, in iprint
    f.write((indent + s).rstrip() + "\n")
UnicodeEncodeError: 'ascii' codec can't encode characters in position 753-755: ordinal not in range(128)```

This must have been introduced recently, because I successfully built python bindings a few weeks ago (using the same `Dockerfile` that produced the compilation error today).  

It would be quite useful to have a tag for the latest commit with working python bindings. However I noticed that the last tags for the repo are a couple of years old.  

Does the project use any continuous integration? It seems like tagging the latest working build would be a relatively easy thing to do with CI.  I would be happy to provide the `Dockerfile` I'm using, although it is more or less bog standard.

rueberger, 2018-11-01:
Regarding the CI question: I have since learned a bit more about how drake does CI. 

The bug was either introduced very recently or caught by CI because the nightly binary works fine.

EricCousineau-TRI, 2018-11-01:
Sorry for not commenting after self-assigning: I'm not sure why you're encountering this error; can you post your Dockerfile, and possibly use a pegged version of Drake (using the `git sha`)?

The Python bindings should be working in Bazel within the [supported configurations](https://drake.mit.edu/developers.html#supported-configurations), as we would see the errors in our Continuous (run on merges to `master`) or Nightly jobs.

On the error specifically: I'm not sure `mkdoc.py` is struggling on your system with that unicode character; can you insert a debug statement to do:
`print(repr(indent + s))`
and let me know what value it's choking on?

jamiesnape, 2018-11-02:
Try something like:
```
apt-get update && apt-get install --no-install-recommends --yes locales && locale-gen en_US.UTF-8 && rm -rf /var/lib/apt/lists/*
```

agalbachicar, 2018-11-02:
It worked! Got the same issue.

gizatt, 2018-11-02:
I can corroborate that we've been seeing this on the Spartan CI (linked
from github.com/RobotLocomotion/spartan, sorry I'm on a phone...) for
around a week, though only in the using-drake-master case. (Spartan's
pegged Drake is fine.) Maybe useful as an additional datapoint?

jamiesnape, 2018-11-02:
It's purely a Docker issue. The images are so minimal that they do not generate the `en_US.UTF-8` locale.

rueberger, 2018-11-02:
Can probably close this then. My observations are explained by the missing `en_US.UTF-8` locale.

EricCousineau-TRI, 2018-11-02:
Ah, that is my bad then; I'll update the docker recipe.

jamiesnape, 2018-11-02:
Maybe we catch `UnicodeEncodeError` and give a better error message too (something like "en_US.UTF-8 locale has not been generated") for the benefit of people rolling their own Docker images.

EricCousineau-TRI, 2018-11-02:
Yup, can do; and perhaps we can add `locale` to `prereqs` too, for simplicity?
(If not, then yeah, we can just keep it in the recipe).

jamiesnape, 2018-11-02:
You could do, but I wouldn't run `locale-gen` other than in the Dockerfile.

EricCousineau-TRI, 2018-11-02:
It doesn't seem to hurt as a no-op, and is effectively a prereq, so I'll try out adding it there.

PR #9918 (2018-11-09): ubuntu: Ensure `locales` is installed for `mkdoc`
Resolves #9903.

For testing, inject `raise UnicodeEncodeError('asdf', u'asdf', 3, 4, 'asdf')` in the `try` / `except` block.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/9918)
closed 2018-11-09 by jwnimmer-tri