RobotLocomotion/drake #9905: Drake needs a mechanism for automatically computing contact forces between overlapping, non-convex bodies

edrumwri, 2018-11-01:
Our present working model is to manually add spheres to the surface of bodies at likely points of contact. FCL allows us to then query the distance between each of these spheres and a mesh (from another body). 

We are working on implementing a solution that computes contact stresses between pseudo-rigid bodies with non-convex geometries.

tri-ltyyu, 2018-11-05:
Completed design phase. Need to come up with plan on how to divide implementation.

tri-ltyyu, 2018-12-03:
Evan and Sean reviewed some of the APIs. Prototyping with MBP.

tri-ltyyu, 2019-03-01:
Prototype API to be reviewed.

sherm1, 2019-03-27:
Status: held design review for ContactSurface query API & data structure using WIP PR #11034.

tri-ltyyu, 2019-04-02:
@edrumwri, what other pieces are ongoing/next as part of this epic? if there're other issues created, could you link them to this please? Thanks!

edrumwri, 2019-04-02:
This epic has begat other epics (leaf-level issues should emerge as one dives through the hyperlinks):

#11034 (PR for the contact surface API, perhaps @DamrongGuoy wants to create the necessary epic issue for tracking progress on the entire process of the computing the contact surface for the hydroelastic model?)
#10809 for visualizing the contact surface
#11057 for solving the resulting stiff ODEs that will emerge from the hydroelastic model

Since #11034 is maturing, I will soon be able to work on extending MBP to compute the forces applied to the rigid bodies as a function of the hydroelastic model (which PR #10767 made possible).

DamrongGuoy, 2019-04-02:
Any tips in creating epic?  I’d like to create one for the entire process
of computing the contact surface in Q2.

-- 
Damrong Guoy, Ph.D. (damrong.guoy@tri.global) (650)673-2605
Toyota Research Institute/ Robotics/ Dynamics, 4440 El Camino Real, Los
Altos, CA 94022, USA

edrumwri, 2019-04-02:
I'd suggest breaking down that large (epic) task into the clear subtasks, of which building the API is one. Perhaps just look through the "hydrostatic implementation document" that Sherm and I wrote and record the tasks listed under geometry?

tri-ltyyu, 2019-04-03:
I agree with breaking down the entire hydrostatic epic into smaller pieces - either phases or the subcomponents (e.g. physics, geometry, etc.). Happy to build the epic and link the issues if you. Regarding #11809, I believe the discussion we had is to leverage RViz - is that your understanding as well?

edrumwri, 2019-04-03:
That was the last recommendation from Calder and Eric, but I've not become
comfortable developing on top of an experimental codebase likely to be in
flux. No decision need be taken quite yet- the contact surface API needs to
land first, at minimum.

--------------------------------
Evan Drumwright
Senior Research Scientist
http://positronicslab.github.io
Toyota Research Institute
Palo Alto, CA

edrumwri, 2019-12-02:
I'm going to mark this one closed, as I think the rolling sphere examples shows that we have completed this one in spirit. The remaining tasks in Spiral I (according to the internal TRI hydroelastic implementation document) are:

1. Add visualization of traction and slip velocity vectors (for debugging purposes)
2. Compute the equilibrium pressure surface (we didn't do this yet because we had some theoretical issues that we have now resolved).

closed 2019-12-02 by edrumwri