RobotLocomotion/drake #9909: Removing geometry from SceneGraph

SeanCurtis-TRI, 2018-11-02:
This is a sub-task that arose from the discussion in #9665. Summary of feature detailed below (discussion of many details in original issue).

What this is not
---------------------
  - Not changing the number of bodies.
  - Not changing the number of frames (in the GeometryFrame/FrameId sense of the word)

What this is
--------------
Proposed API:

```c++
// Note the first argument is the *mutable* Context for the SceneGraph.
GeometryId RegisterGeometry(Context<T>* context, SourceId source_id, FrameId frame_id,
                              std::unique_ptr<GeometryInstance> geometry);

GeometryId RemoveGeometry(Context<T>* context, SourceId source_id, GeometryId geometry_id);

void ExcludeCollisionsBetween(Context<T>* context, const GeometrySet& setA, 
                              const GeometrySet& setB);
```

Details and implications

- The source id for registering the geometry must match the source id of the frame on which it is registered. For the expected use case, that is the source id that the MBP has acquired. So, ask the MBP for its source id to use here. (The exception would be the world frame.)
- The FrameId will likewise be acquired from the MBP. Based on the body, ask MBP for the corresponding geometry::FrameId. If it's the world frame, you can also just directly ask SceneGraph::world_frame_id(). However, both MBP and SG should give you the same value.
- The context will be modified in place.
Changing the geometries affixed to a registered frame will not have any impact on the connections into SceneGraph (the FramePoseVector instances in the abstract ports.)
- After registering a new geometry, collision filtering can be applied to the new geometry and pre-existing geometries via SceneGraph::ExcludeCollisionsBetween(). I'd recommend pre-building a GeometrySet containing all of the geometries you'd generally want to filter w.r.t. the dynamic geometry (I assume it's largely constant as geometries come in and out).
- *Eventually* the functionality in `SceneGraph` responsible for configuring geometry during System construction will *all* be overloaded with a "Modify the given mutable Context" variation. While resolving this issue, it is currently undecided what degree this will happen.

siyuanfeng-tri, 2018-11-02:
From the user end, it would be ideal to have this and @hongkai-dai 's new MBP + SG IK backend around **mid Nov**.

SeanCurtis-TRI, 2018-11-14:
The solution will take two PRs.

The first PR (#10029) adds the API for removing geometry from the `SceneGraph` model.
The second PR (no number yet), will add the API variant that accepts a Context and modifies the state in the context instead of the model.

SeanCurtis-TRI, 2018-11-29:
PRs are merged to master. Feature is now available.

/cc @siyuanfeng-tri @hongkai-dai @avalenzu @calderpg-tri

closed 2018-11-29 by SeanCurtis-TRI