RobotLocomotion/drake #9912: Badly formatted `@throws` doxygen throughout geometry

SeanCurtis-TRI, 2018-11-02:
There are numerous examples in the `geometry/` folder where the conditions for throwing exceptions are documented as:

```c++
/**
...
 @throws std::logic_error  1. If the `source_id` does _not_ map to a
                             registered source,
                             2. If the `parent_id` does _not_ map to a known
                             frame or does not belong to the source, or
                             3. a context has been allocated.  */
```

This renders badly.

[image]

These need to be cleaned up so that they render reasonably.

(TBD Exactly what the "best" alternative is.)

EricCousineau-TRI, 2018-11-02:
@jamiesnape may have some suggestions.

sherm1, 2018-11-02:
The way you formatted in PR #9813 came out beautifully in doxygen:
[image]

That was
```
   @throws std::logic_error
                            1. if the group name is invalid,
                            2. if the property name is invalid, or
                            3. if the property type is not that specified.
```

jamiesnape, 2018-11-02:
You need a line break (and optional whitespace) before each list item. So in the original example you need a line break after std::logic_error.

This would work:
```
  @throws std::logic_error
                          1. If the `source_id` does _not_ map to a
                             registered source,
                          2. If the `parent_id` does _not_ map to a known
                             frame or does not belong to the source, or
                          3. a context has been allocated.
*/
```

jamiesnape, 2018-11-02:
I should add though, that will render horribly in the Python documentation. reStructuredText requires a line break before a list, but that would separate the list from the `@throws` tag in Doxygen as well as in the Python documentation generation script.

SeanCurtis-TRI, 2018-11-05:
That's the conclusion I'm coming to; the enumerated list just doesn't seem to be compatible with the multitude of media through which it is to be consumed.  Siiiigh.

SeanCurtis-TRI, 2019-03-22:
Note: PR #10999 resolves this for `scene_graph.h`; the rest of geometry still needs to be cleaned up.

tehbelinda, 2019-10-08:
Remainder addressed by #12156, let me know if there's anything I missed

closed 2019-10-08 by tehbelinda