RobotLocomotion/drake #9916: Mac: Ensure `numpy` and `nlopt` do not conflict

EricCousineau-TRI, 2018-11-02:
Tracking a known an issue for Mac, shown in TravisCI in `drake-shambhala`.

Example failure: [link](https://travis-ci.org/RobotLocomotion/drake-shambhala/jobs/449540487#L970-L972), snippet:
```
Installing nlopt
Error: numpy 1.15.3 already installed
Installing numpy has failed!
```

I believe @jamiesnape's possible solution is to provide an alternative formula for `nlopt` that does not include Python bindings.

jamiesnape, 2018-11-02:
It is a little more subtle than that, but nevertheless the alternative formula would be https://github.com/RobotLocomotion/homebrew-director/blob/master/Formula/nlopt%402.4.rb.

jamiesnape, 2019-07-05:
Closed by https://github.com/Homebrew/homebrew-core/pull/39620.

closed 2019-07-05 by jamiesnape