RobotLocomotion/drake #9917: Failure in Diagram<double>::ToAutoDiff()

hongkai-dai, 2018-11-02:
I have a test that I first constructs a `Diagram<double>`, and then call `Diagram<double>::ToAutoDiff()`. The transmogrofication fails in `PortsAreValid()` check. The code in is my branch https://github.com/hongkai-dai/drake/tree/diagram_to_autodiff_error. To reproduce the bug, do
```
$ bazel test --compilation_mode=dbg //multibody/inverse_kinematics:minimal_distance_constraint_test
```

The failure is that during transmogrification 

https://github.com/hongkai-dai/drake/blob/aa0334af58524149cb8c8987870c6d8e36637e16/multibody/inverse_kinematics/test/minimal_distance_constraint_test.cc#L12

 the blueprint seems to lose the ports. In https://github.com/hongkai-dai/drake/blob/aa0334af58524149cb8c8987870c6d8e36637e16/systems/framework/diagram.h#L1531, it thinks `dest.first->get_num_input_ports() = 0`.

I think there is a bug in the code, since when I construct `Diagram<double>` in https://github.com/hongkai-dai/drake/blob/aa0334af58524149cb8c8987870c6d8e36637e16/multibody/inverse_kinematics/test/inverse_kinematics_test_utilities.cc#L84, it also checks  `PortsAreValid()`, and the check passes for this `Diagram<double>`. The error only happens when I transmogrify `Diagram<double>` to `Diagram<AutoDiffXd>`.

@sherm1 @jwnimmer-tri

jwnimmer-tri, 2018-11-02:
Another hypothesis is that one of the individual systems _within_ the diagram has a different number of ports after transmogrification.  I believe, in the Diagram code, we assume that the shape of a system is invariant across transmogrification.  (We should probably check for that explicitly and throw a useful diagnostic if the assumption is violated.)

sherm1, 2018-11-03:
@hongkai-dai I think I have a quick workaround for you. Add the line
```c++
DeclareSceneGraphPorts();
```
to MultibodyPlant's scalar-conversion constructor right before `FinalizePlantOnly()` [here](https://github.com/RobotLocomotion/drake/blob/master/multibody/multibody_tree/multibody_plant/multibody_plant.h#L233).

Meanwhile I'll work up a PR that can catch these problems earlier.

sherm1, 2018-11-03:
(I'm seeing some other MBPlant unit test failures now so this is likely not enough -- but it may be enough for your tests.)

hongkai-dai, 2018-11-03:
Thanks @sherm1 and @jwnimmer-tri for looking into it! I think maybe next week we can discuss with Alejandro. I worked on the PR #9824 so as to expose the pain points. I think there are two pain points
1. Missing analytical gradient.
2. Error in transmogrification.

If we can have the analytical gradient, then we do not need the transmogrification. So I think it is worthwhile to have a discussion with Alejandro also, to get a schedule on which issue should be addressed first.

amcastro-tri, 2018-11-07:
@hongkai-dai, analytical Jacobians is on the top of my list. I was working on it before leaving on vacation and I'll resume ASAP.

hongkai-dai, 2018-11-07:
Thanks @amcastro-tri for pushing on it!

PR #9928 (2018-11-08): Include missing SceneGraph ports when transmogrifying MultibodyPlant
The scalar converting constructor (aka transmogrifier) in MBPlant was not reconstructing the SceneGraph ports for the new MBPlant.

Still needs:
- a unit test for an MBPlant/SceneGraph diagram transmogrification
- a framework addition that detects if the result of a transmogrification doesn't have the same topology as the original

@hongkai-dai please check that this fixes #9917.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/9928)
closed 2018-11-08 by amcastro-tri