RobotLocomotion/drake #9929: Simulator's event collections carry hidden state

RussTedrake, 2018-11-05:
The `Context` is supposed to contain everything required to deterministically reproduce a simulation.  I believe that the current implementation of the event handling violates that contract.

Effectively, the variables declared here: https://github.com/RobotLocomotion/drake/blob/master/systems/analysis/simulator.h#L574
act like hidden state.  Shouldn't the event collections be stored in the Context?

As just one example of this pitfall, it looks like if I had a `LeafSystem` that implemented `DoCalcNextUpdate` (which takes a mutable event collection), with something like "if (t=0.5) then declare a timed event at t=2".  I believe that if I started the simulator at t=0, and called
``` 
simulator.StepTo(5.0);
```
I would see that event (assuming I got a step at 0.5), but if I instead called
```
simulator.StepTo(1.0);
simulator.StepTo(5.0);
```
then I would drop the event at the end of the first `StepTo` and not handle it in the second call to `StepTo`?

N.B. -- I was also surprised to see code like this: https://github.com/RobotLocomotion/drake/blob/master/systems/lcm/lcm_log_playback_system.cc#L44 ,
where the time of an `Event` with `TriggerType == kTimed` is passed back as a return value of this method instead of staying associated with the event as `EventData`?   What would you do if a `LeafSystem` added multiple events to the collection during that call?

sherm1, 2018-11-05:
It is OK for the Simulator to have its _own_ "state" (it is not part of the System), but that should just represent its own internal progress, not that of the System. The example you gave is of a broken `CalcNextUpdateTime()` method -- its behavior is dependent on the Simulator's sample times, which are non-physical and permitted to vary arbitrarily. (The documentation doesn't mention that, though!)

`CalcNextUpdateTime()` is called with the current time & state (Context) and is responsible for returning when the next update should occur, _assuming_ there will be no relevant state changes before that time. There is no need to save any of its outputs that go unused -- they are commonly thrown away, especially with variables step sizes. The update times must be recomputed with every call because they may change as a result of state changes. The results of a given call are not events, they are just possible triggers to watch out for as time advances, subject to later revision as more information becomes available.

If you want to trigger a timer (for example) as a result of some event e, the handler for e should set a discrete or abstract state variable when e occurs -- for example, store the time tₑ of e's occurrence. Subsequent calls to CalcNextUpdateTime() will see that in the input Context and can report that the next update time is tₑ+100ms (assuming that the current time is before that).

I'm not sure I follow your question about the logging code. I didn't see anything obviously wrong with it.

> What would you do if a LeafSystem added multiple events to the collection during that call?

The earliest-scheduled of those would determine the returned next update time, and if several were to occur at the same time they would be passed to the handlers together. EventData can be used to communicate with the handlers if necessary to distinguish simultaneous events (we do include the period and offset for periodic events) but that is independent of the return value of CalcNextUpdateTime().

@edrumwri may have more to say about this.

edrumwri, 2018-11-05:
Well said, @sherm1. 

Just to expand at a point:

> The example you gave is of a broken CalcNextUpdateTime() method

Right- if you're looking for t=0.5 for you can schedule an event at t=2, you'd just want to express that as `if t _>= 0.5, next event is at 2.0`. 

And to add a broader point:

The stepping function in Simulator was indeed designed to keep the context consistent between calls and the simulation repeatable from the same context.

RussTedrake, 2018-11-05:
> The example you gave is of a broken CalcNextUpdateTime() method -- its behavior is dependent on the Simulator's sample times, which are non-physical and permitted to vary arbitrarily. (The documentation doesn't mention that, though!)

I do think that having some overview documentation would help.  (Sorry if I've missed it).

The doc here: https://drake.mit.edu/doxygen_cxx/classdrake_1_1systems_1_1_event.html#details was helpful - it helped me understand the difference between `PublishEvent`, `DiscreteUpdateEvent`, and `UnrestrictedUpdateEvent`.   The concept that I cannot find anywhere, and is hard to understand from reading the code, is *when* and *how* events can be declared.  I *think* that the answer is that *almost* all events must be declared before the beginning of the simulation -- since they are owned by the `System`, not the `Context`.  The `CalcNextUpdateTime` was the only example I found of passing a mutable `EventCollection` into a `System` method.  (are there others?)

Assuming my understanding is correct, I didn't see any check/constraints on the types of events I could add in `CalcNextUpdateTime`.  Could I add a new witness event, for instance?  I'm guessing you would also call that bad behavior?  If it's not supported, can we document it and even guard against it in the code? (e.g. only trigger types `kTimed` are allowed in the `timed_events` collection?)

RussTedrake, 2018-11-05:
> It is OK for the Simulator to have its own "state" (it is not part of the System), but that should just represent its own internal progress, not that of the System. 

I also think that this needs to be spelled out more carefully (at least for me).  Presumably, we would expect that the result of `StepTo(5.0)` should not change if I called `StepTo(1.0)` first, correct?

We also had the idea of serializing the `Context` (e.g. to disk), and then restoring it for a future playback.  Simulator "state" complicates that thinking, too.  Even the simulator parameters (e.g. `publish_every_time_step`) would need to be restored to have an exact match.

edrumwri, 2018-11-05:
> The concept that I cannot find anywhere, and is hard to understand from reading the code, is _when_ and _how_ events can be declared. I _think_ that the answer is that _almost_ all events must be [...]

Right.

> The `CalcNextUpdateTime` was the only example I found of passing a mutable `EventCollection` into a `System` method. (are there others?)

Right- it's the only one. 

> Assuming my understanding is correct, I didn't see any check/constraints on the types of events I could add in `CalcNextUpdateTime`. Could I add a new witness event, for instance? I'm guessing you [...]

We'd refer to what you mean by "types of events" as types of triggers. We have a "soft guard" against that- the user creates the event but does not set the trigger (the ability to set the trigger is hidden in Doxygen). We do- Simulator or LeafSystem sets the trigger.

> Presumably, we would expect that the result of StepTo(5.0) should not change if I called StepTo(1.0) first, correct?

Right, that's been a driving design requirement.

sherm1, 2018-11-05:
>> The CalcNextUpdateTime was the only example I found of passing a mutable EventCollection into a System method. (are there others?)

> Right- it's the only one.

What about GetInitializationEvents() and GetPerStepEvents() ?

sherm1, 2018-11-05:
>> Presumably, we would expect that the result of StepTo(5.0) should not change if I called StepTo(1.0) first, correct?

> Right, that's been a driving design requirement.

A small caveat: StepTo(1) is guaranteed to have a time step that ends exactly at 1; StepTo(5) is not under that obligation. So you can expect small numerical differences in the final result, below the accuracy threshold for an accuracy-controlled simulation.

sherm1, 2018-11-28:
See #6456 for an extensive list of what needs documenting (thanks, @SeanCurtis-TRI).

PR #12501 (2020-01-09): Adds high level event documentation.
Resolves #9929.
Resolves #6456.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/12501)
closed 2020-01-09 by jwnimmer-tri