RobotLocomotion/drake #9930: Building Systems around a const MultibodyPlant<T>& reference is inconsistent with scalar conversion

RussTedrake, 2018-11-05:
We have a new pattern that is emerging quickly throughout the code (e.g. with `[InverseDynamicsController](https://github.com/RobotLocomotion/drake/blob/master/systems/controllers/inverse_dynamics_controller.h#L82)`) where a system takes a `const MultibodyPlant<T>&` in the constructor, warns the user that the reference must stay valid throughout the lifetime of the controller system, and then uses it for the internal computation.  How would one write a scalar-converting copy constructor for that control system?  I just want to make sure that the design is not fundamentally incompatible with our type conversion.

Per initial slack discussion, @sherm1 said 

> I do think that scheme is incompatible with scalar conversion since there would not be a way to convert that reference to a `MultibodyPlant<U>&`. Inter-subsystem references should be done through ports, which can be reconnected correctly by the containing Diagram's scalar conversion code. A possible compromise would be to store the SubsystemIndex of the MBPlant rather than a pointer to it (assuming a common parent Diagram). That index is preserved across scalar conversion.

Promoting this to an issue to track it.

RussTedrake, 2018-11-25:
fwiw -- i think that the current recommended usage of SceneGraph (on the consumer side), also instructs people into this pattern?  https://github.com/RobotLocomotion/drake/blob/master/geometry/scene_graph.h#L113

fyi -- @SeanCurtis-TRI .

jwnimmer-tri, 2020-12-09:
If the passed-in plant is required to be standalone (i.e., no inputs are connected), then I think the scalar-converting copy constructor for `InverseDynamicsController` could simply scalar-convert the given `plant` reference directly, now keeping the owned copy internally.