RobotLocomotion/drake #9939: mbp: Errors about weld joints should provide more information

EricCousineau-TRI, 2018-11-05:
Ran into a case where an IIWA SDF had a world-weld joint, but we accidentally added a secondary / redundant weld.

The error message from `MultibodyTreeTopolgoy` is verbose, but doesn't point to the frames in question (which is honestly the most useful information).

The easiest solution is to print out the frame index, and instruct the user to use `get_frame(index).name()` to figure out the user-friendly bits.

EricCousineau-TRI, 2020-11-05:
@hongkai-dai ran into this recently:
Slack: https://drakedevelopers.slack.com/archives/C2WBPQDB7/p1604601033080200

Error message:
```
    plant.Finalize()
RuntimeError: This multibody tree already has a mobilizer connecting these two frames. More than one mobilizer between two frames is not allowed
```
(less verbose, but still a bit obtuse for world welding / parent-child mismatch?)

I won't be able to address this - @amcastro-tri I'm assigning to ya, but if you don't think it's actionable, feel free to close!

sherm1, 2020-11-06:
Related: #13040

EricCousineau-TRI, 2021-05-03:
Running into this again in Anzu; in this case, doing a dual-arm setup, and I'm playing with how I glue stuff together (topology) to make calibration (touch-point) a bit simpler for readers to digest.

\cc @siyuanfeng-tri

amcastro-tri, 2021-05-03:
Thanks Eric. cc'ing @joemasterjohn since eventually we'll clean up the whole tree vs model topology.

sherm1, 2022-06-21:
Closing this -- #17429 is close enough.

jwnimmer-tri, 2022-06-21:
Eric's original post (and the follow-up posts) were about adding _two_ welds, where the complaint is that the second weld is redundant.  #17429 is about welding to the middle of a model, which is a different exception path.  They don't seem like the same issue to me?

sherm1, 2022-06-21:
My thought is that if we can properly handle the analysis that leads us to use a reverse mobilizer to fix #17429 that analysis will also detect cases where we can't. But I think you're right that we should leave this open until we can verify that the error messages are tolerable.

PR #22955 (2025-05-01): [multibody] Add tests to show issues 9939 & 17429 are closed
PR #22949 (now allowing reversed weld, revolute, and prismatic mobilizers) fixed long-standing issue #17429. The even-older issue #9939 (bad error message, somewhat related) is also resolved (sometime earlier).

This PR adds unit tests to prove that those two issues are resolved, and closes them.

Closes #9939
Closes #17429

- - -
This change is [[image]](https://reviewable.io/reviews/RobotLocomotion/drake/22955)
closed 2025-05-01 by sherm1