RobotLocomotion/drake #9946: lcm_logger compiler warning produces noise during compilation

RussTedrake, 2018-11-06:
```
INFO: From Compiling external/lcm/lcm-logger/lcm_logger.c:
external/lcm/lcm-logger/lcm_logger.c: In function 'open_logfile':
external/lcm/lcm-logger/lcm_logger.c:150:64: warning: '%02d' directive output may be truncated writing between 2 and 11 bytes into a region of size between 0 and 4095 [-Wformat-truncation=]
             snprintf(logger->fname, sizeof(logger->fname), "%s.%02d", logger->fname_prefix,
                                                                ^~~~
In file included from /usr/include/stdio.h:862:0,
                 from external/lcm/lcm-logger/lcm_logger.c:4:
/usr/include/x86_64-linux-gnu/bits/stdio2.h:64:10: note: '__builtin_snprintf' output between 4 and 4108 bytes into a destination of size 4096
   return __builtin___snprintf_chk (__s, __n, __USE_FORTIFY_LEVEL - 1,
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
        __bos (__s), __fmt, __va_arg_pack ());
        ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
external/lcm/lcm-logger/lcm_logger.c:155:59: warning: '.0' directive output may be truncated writing 2 bytes into a region of size between 1 and 4096 [-Wformat-truncation=]
         snprintf(logger->fname, sizeof(logger->fname), "%s.0", logger->fname_prefix);
                                                           ^~
In file included from /usr/include/stdio.h:862:0,
                 from external/lcm/lcm-logger/lcm_logger.c:4:
/usr/include/x86_64-linux-gnu/bits/stdio2.h:64:10: note: '__builtin_snprintf' output between 3 and 4098 bytes into a destination of size 4096
   return __builtin___snprintf_chk (__s, __n, __USE_FORTIFY_LEVEL - 1,
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
        __bos (__s), __fmt, __va_arg_pack ());
        ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```

RussTedrake, 2018-12-21:
am I the only one that sees this always?  @jwnimmer-tri ?

jwnimmer-tri, 2018-12-21:
Yes, we always see this (on Bionic only), but it only shows up when `@lcm` is re-compiled, which for many people is not often because its object code is always in the Bazel `disk_cache`.

My guess was that this warning does not have a cmdline flag to disable it.  (At least, none is suggested in the diagnostic, and IIRC my web searching didn't find anything.)  If you want to try editing `LCM_COPTS = []` in `tools/workspace/lcm/package.BUILD.bazel` to suppress this, try adding `-w` there and see if that mutes it.  If so, that means there's _some_ flag that could mute it.

If there is no suppression flag, then a resolution would require code changes PR'd to upstream.

PR #10363 (2019-01-09): workspace/lcm: Disable all compiler warnings
Without this, on gcc-7.3 we have `lcm_logger.c:150:64: warning: '%02d' directive output may be truncated writing between 2 and 11 bytes ...`.

The toggle -Wno-format-truncation is not supported by clang, so it's easiest to just disable everything.

Closes #9946.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/10363)
closed 2019-01-09 by jwnimmer-tri