RobotLocomotion/drake #9948: xcode clang complains about unused argument `-pie` on every linking operation

RussTedrake, 2018-11-06:
```
INFO: From Linking systems/sensors/image_test:
clang: warning: argument unused during compilation: '-pie' [-Wunused-command-line-argument]
INFO: From Linking examples/bouncing_ball/bouncing_ball_test:
clang: warning: argument unused during compilation: '-pie' [-Wunused-command-line-argument]
INFO: From Linking examples/simple_discrete_time_system:
clang: warning: argument unused during compilation: '-pie' [-Wunused-command-line-argument]
INFO: From Linking systems/sensors/lcm_image_array_receive_example:
clang: warning: argument unused during compilation: '-pie' [-Wunused-command-line-argument]
INFO: From Linking systems/sensors/depth_sensor_to_lcm_point_cloud_message_test:
clang: warning: argument unused during compilation: '-pie' [-Wunused-command-line-argument]
INFO: From Linking examples/simple_continuous_time_system:
clang: warning: argument unused during compilation: '-pie' [-Wunused-command-line-argument]
```

jamiesnape, 2018-11-06:
https://github.com/bazelbuild/bazel/pull/6478, but I have very little confidence they will ever review it.

jamiesnape, 2018-11-06:
If it is getting really annoying, the sledgehammer approach is to locally add 
```
build --linkopt=-Wno-unused-command-line-argument
```
to `~/.bazelrc` or `%workspace%/user.bazelrc`.

RussTedrake, 2018-11-06:
no rush (marked as low priority).  thanks for being on it and tracking it.

EricCousineau-TRI, 2019-01-10:
Updated to Bazel 0.21.0, which incorporated https://github.com/bazelbuild/bazel/pull/6478

closed 2019-01-10 by EricCousineau-TRI