RobotLocomotion/drake #9956: drake's meshcat-server can't actually open a display

RussTedrake, 2018-11-07:
Quickly diagnosed by @gizatt and @pangtao22 ...

The symptom:  Running
```
bazel run @meshcat_python//:meshcat-server
```
provides the standard output, with a link to a web url.  But opening that url results in 404 errors.

@gizatt suspects that the problem is that https://github.com/RobotLocomotion/drake/blob/e5cad7b9be04bdeeaa4824f4150e9911bb55fb93/tools/workspace/meshcat_python/package.BUILD.bazel
does not install meshcat's [src/meshcat](https://github.com/rdeits/meshcat-python/tree/master/src/meshcat)/viewer subdirectory, which contains all the HTML / JS files required for the webserver to actually serve the 3D visualizer in the browser.

Low priority only because pip installing meshcat still provides a working meshcat-server that gives a workaround.

jamiesnape, 2018-11-07:
Ok, it is a quick fix if that is the case.

RussTedrake, 2018-11-11:
Resolved by #9979 and #9982

closed 2018-11-11 by RussTedrake