RobotLocomotion/drake #9976: Port systems/analysis integrator tests to MBP

jwnimmer-tri, 2018-11-08:
There are two programs remaining under `//systems//analysis/...` that use `RigidBodyTree`:
```
//systems/analysis:runge_kutta3_integrator_test
//systems/analysis:semi_explicit_euler_integrator_test
```

I assume that the right answer is to port these to MultibodyPlant, is that correct?

Relates #9783.

sherm1, 2018-11-08:
Yes! I think this one is for you, @edrumwri. Would be nice to get these RBPlant dependencies out of our own code now that MBP can do the job.

jwnimmer-tri, 2018-12-17:
As of today, there are only three non-attic things in Drake that depend on attic code.  All three are unit tests, and two of the three are these unit tests.  I plan to have the third item reworked to MBP before the break, so: heads-up that this issue will be the _only_ RBT code outside of the attic pretty soon.