RobotLocomotion/drake #9983: Update docs to MakeFromOrthonormalColumns()/MakeFromOrthonormalRows().

amcastro-tri, 2018-11-09:
Can I make a request here?. The docs for this method say:
>  The rows of `R_AB` are `Ax`, `Ay`, `Az` whereas the columns of `R_AB` are `Bx`, `By`, `Bz`.

I think we'd like saying:

>  The rows of `R_AB` are the versors for base A `Ax_B`, `Ay_B`, `Az_B` expressed in B whereas the columns of `R_AB` are the versors for base B `Bx_A`, `By_A`, `Bz_A` expressed in frame A.

Most importantly I'd like the docs to convey the "expressed in" idea. Similarly for `MakeFromOrthonormalColumns()`.

sherm1, 2018-11-09:
TIL "versor" which apparently means "unit vector in the direction of some vector". Is that a common term? If not we might want to stick with "unit vector".

amcastro-tri, 2018-11-09:
It is but "unit vector" works just as well. The most important point in this docs is that we cannot say that, for instance, that Ax is the row of R_AB. Ax is a mathematical objects, only its components can be part elements of matrix R_AB and that is why I think it's important saying something more like Ax_B, specifying the frame. I might be missing something, but I always think of `R_AB` as a matrix where I hang the column vectors containing B's versors expressed in frame A.

sherm1, 2018-11-09:
Agreed!

mitiguy, 2018-11-10:
This is being addressed as part of PR# 9988.

closed 2018-11-13 by mitiguy