RobotLocomotion/drake #9992: _DeclareInitializationEvent requires _DoPublish to call LeafSystem._DoPublish

RussTedrake, 2018-11-10:
Just lost quite a bit of time (print debugging my way through the event stack) diagnosing why my initialization event was not working in:
https://github.com/RobotLocomotion/drake/pull/9982

Not sure how best to address it.  But this is a hazard for python LeafSystem authors.

sherm1, 2018-11-10:
There are two methods for providing event handlers: 
- provide a callback associated with an event, or
- override the whole event dispatcher (e.g. DoPublish()) and put code there directly.

The example in #9982 mixes both of those, which is likely why it is confusing. It would be cleaner just to use a callback for both the "periodic publish" and "initialize" messages, so that the LeafSystem default dispatcher could process them both. Then you wouldn't need to override `DoPublish()`. (Alternatively, the `DoPublish()` override could look in the provided event list and perform the dispatching itself based on the event types -- but I think the callback approach is nicer.)

RussTedrake, 2018-11-10:
The only event that I can currently declare as an initialize event from python has type publishevent.  When the Initialize() routine in Simulator goes to handle that event, it calls LeafSysem.DoPublish to do the work.  If you have (separately) overloaded that function, and not handled the initialize event type, then the handler never gets called.

So the declareinitializationevent method has a hidden dependence on the DoPublish implementation.

sherm1, 2018-11-10:
Generally I think it is not a good idea to overload DoPublish at all. We got in the habit of doing that before events had their own callbacks. If both the periodic and initialization events were declared with callbacks, you wouldn't have had to overload DoPublish, and the default implementation would have done what you want. 

The only good reason remaining to overload DoPublish is if you want to handle simultaneous events in some non-standard manner (maybe by prioritizing them, or eliminating duplicates). Otherwise the default implementation (which runs through all the events it gets and invokes each of their callbacks in turn) should be fine.

I'm not claiming the documentation makes any of this clear! #9929 still applies. But that is the direction I think we should encourage people to go -- always provide a callback with every event; never overload DoPublish. I'm not sure yet that the API makes this smooth.

RussTedrake, 2018-11-11:
I did not realize that was the new recommendation.  I would be happy to update the examples I touch to that style, but will need the python bindings for DeclarePeriodicEvent (including the version that takes the event).

EricCousineau-TRI, 2018-11-12:
Given that the high-level takeaway I get from this is to discourage users from overriding `DoPublish`, and use callbacks, is there a way that we can make that clearer in the short term, pending a broader resolution of #9929?

I can do a follow-up PR to #10004 that changes the unittests / examples to not override `_DoPublish` but instead use the event interface.

sherm1, 2018-11-12:
@EricCousineau-TRI that would be great -- I think modifying the pedagogical examples is the best way to get people to use the new pattern. Even once we get #9929 documentation written, I suspect most users will copy-paste from examples rather than wade through the docs.

PR #10004 (2018-11-12): Bind PerStep and Periodic event methods to support callback-using API
Allow per-step and periodic events to be specified using callbacks in Python (like initialization events).

In practice, these can be used to avoid overriding DoPublish() and make for more-straightforward examples.

Resolves #9992.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/10004)
closed 2018-11-12 by sherm1