RobotLocomotion/drake #10000: json: Archive of v3.2.0 is 110M

EricCousineau-TRI, 2018-11-12:
When building on my laptop at home, I noticed a large download for `json`, and just tried downloading the archive:
```
wget https://github.com/nlohmann/json/archive/v3.2.0.tar.gz
```

Ends up, this is ~110MB compressed; uncompressed, a majority of the data resides in `/test` and `/benchmark`, neither of which seem important to us. If you strip out `/doc` as well, the resulting uncompressed files are 1.6MB.

It'd be nice to get a version of the code that doesn't have that much uneeded bloat, but I don't know of any simple solutions aside from forking (yuck).

\cc @jwnimmer-tri @jamiesnape

SeanCurtis-TRI, 2018-11-15:
For the record, today I spent 15 minutes downloading json. This is despite the fact that speedtest.net reported download speeds from my computer at 500 Mbps. If the server is being slow, then the 110 Mb is punishing.

jwnimmer-tri, 2018-11-15:
Well, the Drake mirrors don't have json v3.4.0 cached yet, so I guess for you github.com was slow today.  More of a #8595 automation problem than a large file problem -- @jamiesnape FYI.

jamiesnape, 2018-11-15:
As much as I would like to be the solution to all slower downloads, the mirrors aren't going to help much unless we make them the primaries, and the download for JSON has always been large (it grew about 800 KB between versions), so not sure why it is now such an issue. And yes, it is mirrored, though CloudFront does need to propagate the packages to its endpoints.

jwnimmer-tri, 2018-11-15:
Ah.  I was thinking that Bazel tried all mirrors concurrently, but saying it out loud that would be silly.

In any case, I wholeheartedly endorse closing this issue as "not even worth the time investment in discussing".

EricCousineau-TRI, 2018-12-10:
Following endorsement; closing.
Can re-open if anyone feels horribly pained by this.

closed 2018-12-10 by EricCousineau-TRI

EricCousineau-TRI, 2019-03-28:
FTR This is effectively captured by this issue (and its duplicates):
https://github.com/nlohmann/json/issues/96

nlohmann, 2020-05-04:
FYI: https://github.com/nlohmann/json/issues/96 is solved now and the next release will exclude test data and will be much smaller (~ 7 MB).

jwnimmer-tri, 2020-05-04:
Thank you!  In the meantime, Drake has dropped its _direct_ use of `nlohmann/json`, but we still have other projects that use it.  \CC @soonho-tri FYI