RobotLocomotion/drake #10015: systems: Add `DispatchInitializeEvents(System)`

EricCousineau-TRI, 2018-11-13:
In two places ([show_model](https://github.com/RobotLocomotion/drake/blob/1e01440576e3efba6b6829dda45977f4208615b2/manipulation/util/show_model.py#L93) and [`pydrake/systems/test/test_util_py.cc](https://github.com/RobotLocomotion/drake/blob/1e01440576e3efba6b6829dda45977f4208615b2/bindings/pydrake/systems/test/test_util_py.cc#L124-L125)), I've used `Simulator.Initializate()` to dispatch initialization events, which seems silly.

It would be nice to add this as a free function (or a method... but meh).

jwnimmer-tri, 2018-11-13:
Why is it silly?  It does what it says on the tin, and is what you wanted to happen?

EricCousineau-TRI, 2018-11-13:
Sorry: It's overkill, because I've called it when I don't really need any other functionality from `Simulator`.

(Updated text.)

jwnimmer-tri, 2018-11-13:
But you _do_ want other things from the simulator -- you want it to do initialization, and then dispatch the `t = t_0` publication event:
```
diagram = builder.Build()
simulator = Simulator(diagram)
simulator.StepTo(0.0)
```

I think its _worse_ to manually call both publishes, don't you?

EricCousineau-TRI, 2018-11-13:
Hm... Perhaps. But it still feels unclean that our semantics are so complex that you have to instantiate a `Simulator` just to call init + discrete publishes, and rely on the `set_publish_at_initialization` to work (which may or may not be sensitive to #9797).

sherm1, 2018-11-13:
FYI @edrumwri is improving and clarifying the relationship between Initialize() and Publish() in #9766; `set_publish_at_initializer()` won't be necessary in most cases after that.

I'm not sure why you want a special method for initialization events, especially if you really aren't initializing a simulation. "Events" are things that happen as time moves forward. Can you explain? And do you mean publish events only -- what if the initialization events are discrete or unrestricted updates?

You can see from
[Simulator::Initialize()](https://github.com/RobotLocomotion/drake/blob/master/systems/analysis/simulator.h#L491) that it takes quite a bit of machinery to process events correctly. Are you sure that "event" is the right abstraction for what you're trying to do?

There is `System::Publish()` which can force a call to DoPublish() whenever you want one. I presume that won't work here for some reason?

jwnimmer-tri, 2018-11-13:
@EricCousineau-TRI I retract my above, and I apologize for the noise.  I've belatedly realized that this feature request is related to events API design, and I promised myself I would stay away from such discussions subsequent to #6050.  It sounds like @sherm1 is on the case now, anyway.

EricCousineau-TRI, 2018-11-13:
The use case is specific to calling publish initialization + discrete publish events from visualization tools, and for general testing.

For visualization, it would be nice to not require a Simulator, as there's far more machinery there than seems necessary. In this case, discrete / unrestricted updates are not necssary.

For debugging, it's nice to have it decoupled, especially for testing simple bits for `T=Expression` in Python; from #8452:
https://github.com/RobotLocomotion/drake/blob/8166ac366137ea0912cc0756902f917eb47ce4e7/bindings/pydrake/systems/test/test_util_py.cc#L144-L145
In this case, I couldn't instantiate `Simulator<Expression>` because it default-constructs an integrator (I believe this failed at linker time...), so I had to break out the initialization events.

(In this case, I also only needed publish initialization events, so perhaps this could be pruned down to just publish @ init.)

sherm1, 2018-11-13:
Reminder -- the event abstraction works like this:
```
                          --> handler1
triggers  -->  dispatcher --> handler2
         Events           --> etc.
```
Triggers include: initialization, per-step, periodic time, specified time, zero crossing (witness), and "forced"
Dispatchers are: publish, discrete update, unrestricted update
Handlers are: callbacks (preferred), or explicit code in overridden dispatchers

Right below the line you mentioned you have this code:
https://github.com/RobotLocomotion/drake/blob/8166ac366137ea0912cc0756902f917eb47ce4e7/bindings/pydrake/systems/test/test_util_py.cc#L148-L151

For a test harness, why not handle initialization events (and per-step events, etc.) similarly? Call the trigger method that generates the event list, then pass that to the appropriate dispatcher.

EricCousineau-TRI, 2020-07-28:
Circling back, didn't get to this, but maybe it's relevant given discussion in #13741? (or maybe not, I dunno)

sherm1, 2020-07-28:
#13741 is specifically about when to call `Simulator::Initialize()`. Here you said you don't want to have to create a Simulator at all. So I think the issues are distinct. If I understand your request here, it is that you would like to trigger and handle just the initialization events that have Publish handlers (no discrete or unrestricted update). If so that can be done with the existing API like this:
```c++
auto events = system.AllocateCompositeEventCollection();
system.GetInitializationEvents(context, events.get());
system.Publish(context, events->get_publish_events());
```

Its not clear to me that this is common enough to require sugar to make it prettier. Can we close this issue?

EricCousineau-TRI, 2020-07-28:
It's used in 2 places in Drake, and 8 places in Anzu.
I'd like the sugar, will submit a short PR.

EricCousineau-TRI, 2020-07-30:
Closing in lieu of #13776.

closed 2020-07-30 by EricCousineau-TRI