RobotLocomotion/drake #10018: texture file documentation is out of date (and out of tree)

RussTedrake, 2018-11-13:
The instructions here: 
https://drake.mit.edu/models.html#texture-file-format
reference a python file convertCollada.py in oh-distro.  

That file doesn't work with our current version of vtk.  But @kmuhlrad says that changing `SetInput` -> `SetInputData` was enough to get it working.  

First question: is this still a recommended workflow?   @SeanCurtis-TRI ?
If yes, second question:  should we pull that file into our source tree and update it?

RussTedrake, 2018-11-13:
Thx.  In addition to drake_visualizer, I also care about the rgbd sensor being able to see the textures.

SeanCurtis-TRI, 2018-11-13:
RGBD sensor currently loads pngs. Limited, but at least no dependent on VTP. :)

RussTedrake, 2018-11-14:
Just found this.  Related: #9287.

jwnimmer-tri, 2020-04-21:
Per #9287 and #10963, it seems like the lying documentation is no longer lying, by virtue of being deleted.  This read to me more like a bug report (lying docs) that a feature request (please document how I accomplish X), so I'll close it now.

closed 2020-04-21 by jwnimmer-tri