RobotLocomotion/drake #10019: python, cmake, gdb: Unable to find debug symbols?

EricCousineau-TRI, 2018-11-14:
@edrumwri was doing some debugging with CMake + Python, and was unable to get debug symbols.

I briefly tried it out on my machine, and also could not get symbols.

My repro steps:
```
$ cd drake
$ mkdir build && cd build
$ cmake .. -DCMAKE_BUILD_TYPE=Debug
# `build/bazel.rc` has `-g` present.
$ make -j
$ export PYTHONPATH=${PWD}/install/lib/python2.7/site-packages:${PYTHONPATH}
$ gdb -ex "dir ${PWD}/.." \
    -ex "set breakpoint pending on" -ex "b drake::FindResourceOrThrow" \
    -ex "run" \
    --args python -c "import pydrake.common as m; m.FindResourceOrThrow('drake/stuff')"
(gdb) bt
#0  0x00007ffff383bdcf in drake::FindResourceOrThrow(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >) ()
   from /home/eacousineau/proj/tri/repo/branches/drake/tmp/drake/build/install/lib/python2.7/site-packages/pydrake/common/../../../../libdrake.so
#1  0x00007ffff6955d85 in ?? () from /home/eacousineau/proj/tri/repo/branches/drake/tmp/drake/build/install/lib/python2.7/site-packages/pydrake/common/_module_py.so
(gdb) list
6       ../Modules/python.c: No such file or directory.
```

Checking the file, it seems there may be no debugging symbols?
```
$ file ./install/lib/libdrake.so 
./install/lib/libdrake.so: ELF 64-bit LSB shared object, x86-64, version 1 (SYSV), dynamically linked, BuildID[sha1]=a4c123cfc8d16e9a079a418cc296da042585de8e, stripped
```

@jamiesnape Can I ask if there is a better workflow for this?

In the meantime, I believe Evan can debug his script using Bazel.

jamiesnape, 2018-11-14:
We strip the binaries when we install via Bazel, so there won't be any symbols. Nothing to do with Python (nor CMake really).

EricCousineau-TRI, 2018-11-14:
Ahhh, gotcha. Mind if I add in an option to `install.py.in`, so `--no-strip`, and somehow enable that depending on the `CMAKE_BUILD_TYPE` (`Debug` or `RelWithDbgInfo`)?

jamiesnape, 2018-11-14:
You can, but you need to be sure to strip out SNOPT symbols regardless of any setting.

jwnimmer-tri, 2018-11-14:
> How do I go about determining which are SNOPT symbols?

There is a test case at https://github.com/RobotLocomotion/drake/blob/master/tools/install/libdrake/test/snopt_visibility_install_test.py that calls out some (but not all) patterns for snopt symbols.

> And is that only important for redistribution?

It's important that Jenkins https://drake-jenkins.csail.mit.edu/view/Packaging/ builds never (even accidentally) upload an install tarball with snopt symbols.  Right now they are all `-release` mode, but you could imagine they could someday be `-debug`, or that the configuration flag setting had a bug that turned on debug builds for `-release` CI.

jwnimmer-tri, 2018-11-14:
Possibly it would be good enough to let people comment out `check_call(['strip', dst_full])` during local development, and call it a day.  Perhaps our time is better spend in documenting a bazel-centric debugging environment for pydrake in from-source builds.  I'm not sure routing through the `install` target should really be a part of a development workflow in any case.

EricCousineau-TRI, 2018-11-14:
Gotcha, that makes sense then.
I'll see if there's a way to very briefly hint that `CMAKE_BUILD_TYPE=Debug` won't produce debug symbols (maybe add a footnote for masochists), and direct people to debug with Bazel.

edrumwri, 2018-11-28:
Just following up on this discussion a little: developing in pydrake is _nice_. The biggest (current) drawback is that debugging into the C++ code is frequently required. We shouldn't require a pydrake developer to build within the Drake dev environment (which includes constructing bazel build files for their python code) in order to debug into the C++ code.

PR #10103 (2018-12-12): install, cmake: Enable not stripping symbols for debug
pydrake: Add debugging info for cmake build

Resolves #10019

Given @edrumwri's prior comments, I think this is a good idea; I've sprinkled warnings all throughout to avoid any potential mishaps for redistribution, as I think it's more useful to keep things simple to enable user debugging workflows as Evan has pointed out.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/10103)
closed 2018-12-12 by EricCousineau-TRI