RobotLocomotion/drake #10022: Manipulation station (cupboard, iiwas, wsg, cameras, and objects) should all specified via sdf

RussTedrake, 2018-11-14:
The manipulation station example should migrate to loading the entire configuration from sdf, with the only assumptions being that there is one IIWA model, one WSG model, and some number of cameras.

Consider also including the camera serial numbers as a tag in the sdf.

EricCousineau-TRI, 2018-11-18:
From having dealt with this, I'm not sure if SDF is the correct solution unless we want to fix our resource paths and do everything through Xacro (which is inflexible since this is going to be compile-time gen, and we should avoid needing this at all during runtime).

Perhaps simple YAML assembly is the right step forward, after which we should focus on having or creating a better format for assembling scenes.

RussTedrake, 2019-03-08:
I believe that the current plan of record for this is to port the yaml parsing from anzu into drake.  @sammy-tri -- perhaps I could put this onto your plate? (priority is marked as low)

jwnimmer-tri, 2019-03-08:
@sammy-tri If you like, I could start moving the visitor/archive pattern over?  That seems to have stabilized by this point, even though the schemas that use it are still evolving.

jwnimmer-tri, 2019-03-08:
And to clarify.. I think the YamlReadArchive is stable enough to move to Drake.  I don't think we can move many specific schemas (like model directives) yet -- they are some mix of too unstable, or too poorly designed, to upstream.

EricCousineau-TRI, 2019-03-08:
+1000 to Jeremy's comment

Extra peanut gallery comment, parallel my other in #10851:
There's been quite a few PRs and issues that (to me, at least) indicate that `ManipulationStation` should be broken apart for applications that do not fit its design (which for simple tutorial uses is *absolutely* great, but over-encapsulated for anything else).

sammy-tri, 2019-03-08:
>  I could start moving the visitor/archive pattern over?

:+1: yeah, I think it's time to do this.  We can discuss further about the specific schemas.

jwnimmer-tri, 2019-03-08:
> I could start moving the visitor/archive pattern over?

=> #10869

huihuaTRI, 2020-02-21:
Just curious about the update on this issue. Another use case of this feature popped up in Drake while trying to simulate HSR. See [slack](https://tri-internal.slack.com/archives/C0U7NABLZ/p1582313399126800) for a brief discussion with Calder.

jwnimmer-tri, 2021-08-10:
As of #13282, the full YAML scene language ("model directives") is available in Drake.  The manipulation_station has not been ported to that yet, though.

RussTedrake, 2021-12-22:
This will be resolved via #14992

closed 2021-12-22 by RussTedrake

jwnimmer-tri, 2023-04-07:
See also https://github.com/RobotLocomotion/drake/issues/19155.