RobotLocomotion/drake #10023: rgb-d cameras can render the 6.881-relevant YCB objects

RussTedrake, 2018-11-14:


RussTedrake, 2018-11-25:
From f2f with @SeanCurtis-TRI:

"for RgbdCamera, it is *definitely* going to look for `my_object.png` for `my_object.obj`. And that is the *only* circumstance under which textures will reach the RgbdCamera.
(emphasize on "only for obj" and "only png".)"

kmuhlrad, 2019-01-08:
In order to run [NVidia DOPE](https://github.com/NVlabs/Deep_Object_Pose) on simulations, we need to render textures through `RgbdCamera`s. This is important for developing clutter-clearing demos on the `ManipulationStation` with perception in both simulation and on real hardware.

SeanCurtis-TRI, 2019-01-08:
I'm introducing a short-term hack that mirrors the old `RgbdRendererVtk` functionality. Given an obj file `/path/to/thing.obj` it would look for `/path/to/thing.png`. If it existed, it would apply it as a texture. This is very hacky, but at least provides a way forward.

In the future, I'll introduce OBJ and MTL parsing and extract textures from there as the most "correct" solution.

We might also explore the possibility of adding tags to the SDF/URDF materials where the `diffuse` vector can be overriden/combined with a `diffuse_map` value. This has lesser value and could be used to supplant any texture indicated in the material library. Generally, this is mucky. But no more mucky than what would ordinarily happen if we specify different diffuse colors for a mesh in the SDF *and* the MTL files.

SeanCurtis-TRI, 2019-01-10:
The `dev/SceneGraph` now applies the same logic as the legacy `RgbdRenderer`: if a mesh is loaded named `/path/to/my/thing.obj`, it will look for a file called `/path/to/my/thing.png`. If it exists and successfully read as a png, it will be assigned as a texture.

The full details are slightly more complicated and this feature will change when it hits mainstream master. Ultimately, we'd like to parse the OBJ, find a reference to an MTL file, parse that and use textures listed there.  Even in this case, there are interesting question about resolving conflicting signals, e.g., what if the SDF file specifies material values (i.e., diffuse) but there is an existing specified material library (mtl) file. What values should be used?

SeanCurtis-TRI, 2019-01-10:
@RussTedrake with the most recent PR into dev/render, should this issue still be open? I can see this either as an issue with current events *or* a reminder to make sure we have this when `dev/render` hits master. I leave it up to you if you want to close this issue in favor of another.

RussTedrake, 2019-01-10:
#10367 looks like the resolution I was hoping for.  Thank you!

closed 2019-01-10 by RussTedrake