RobotLocomotion/drake #10024: Include YCB objects in drake examples

RussTedrake, 2018-11-14:
this involves resolving the licensing issues first.
then curating the model files to our preferred formats. (related to #10023)
then deciding if they live in drake or in a separate model repository.  (e.g. as prototyped in a manipulation station dev branch)

tri-ltyyu, 2018-12-03:
Got email permission from Berk Calli to use dataset. However, since there're multiple schools involved, Corporate Counsel advised license should be indicated on website. Berk is working on this.

tri-ltyyu, 2018-12-08:
License info has been listed on the download website http://ycb-benchmarks.s3-website-us-east-1.amazonaws.com/
Can be used with proper attribution

EricCousineau-TRI, 2019-01-29:
Spoke with @kmuhlrad on Slack about #10518:
The scope for these models is for use in `examples/manipulation_station` for demos involving perceiving those objects. They will be used in a non-trivial fashion in Drake (so Drake will not just be file storage for downstream projects), and there is no plan to increase these number of objects w.r.t. this current use case (manipulation station demos).

Since there are some constraints on how to pull in the data, I'm going to open a replacement PR, add a `README` and ensure we add in the original models per the license they have.

If/when more objects are needed, #6124 (or a replacement issue) should be resolved.

\cc @jwnimmer-tri

EricCousineau-TRI, 2019-01-29:
Er, after looking at #10518, one thing that wasn't discussed in Slack was the size of the PNG files: Each of them are ~8MB of size. While this is kinda subjective, I just don't feel comfortable loading more than 10MB into a PR, especially if we have to commit the originals for legal reasons (see [Slack convo](https://drakedevelopers.slack.com/archives/C0JNB2E3S/p1548453364046800?thread_ts=1548452400.045700&cid=C0JNB2E3S)).

New plan, per Jeremy's prior point:
- [ ] In RobotLocomotion/models, I'm going to rename `master` to `incorrect_master`, and force push a fresh new `master`.
- [ ] Jeremy will set up RobotLocomotion/models with nominal Reviewable permissions + push access, such that review is required (licenses can be reviewed).
- [ ] We'll put the models in there, and make a Drake PR that uses the fresh commit to pull in data.

@kmuhlrad @RussTedrake Does that work for y'all?

kmuhlrad, 2019-01-29:
SGTM

RussTedrake, 2019-01-30:
I am suggesting that we restart that repository with a single commit bringing in the ycb.   No history.

jwnimmer-tri, 2019-01-30:
If / when we click the "delete branch" button on `old_master` from the GitHub web UI, we'll end up with the same result.  The rename just lets us separate that step from moving forward with new-`master` PRs and such.  It lets us look at the prior work in case we want to re-PR it to new-`master`, for example.

RussTedrake, 2019-01-30:
Ok.  Just wanted to make sure we’ll end up with a clean history in the end.

kmuhlrad, 2019-02-04:
Any updates on https://github.com/RobotLocomotion/models/pull/5?

EricCousineau-TRI, 2019-02-04:
Sorry, haven't had time to circle back yet. Is Wed-ish an OK time for it to land?

kmuhlrad, 2019-02-04:
Yep, Wednesday is fine, I just wanted to make sure this didn't get lost.

PR #10532 (2019-02-11): models: Add YCB subset
Resolves #10024

Supersedes #10518

Requires:
- [x] https://github.com/RobotLocomotion/models/pull/5

\cc @kmuhlrad @RussTedrake @jwnimmer-tri 

Prolly should wait for upstream stuff to get figured out before reviewing.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/10532)
closed 2019-02-11 by EricCousineau-TRI