RobotLocomotion/drake #10040: show_image.py complains when image_array_t.header.frame_name=""

RussTedrake, 2018-11-15:
e.g. with
```
LcmImageArraySubscriber: For image channel '{}' (DRAKE_RGBD_CAMERA_IMAGES images), with frames 3:
  Did not find 'camera0' in message
  Did not find 'camera1' in message
  Did not find 'camera2' in message
LcmImageArraySubscriber: For image channel '{}' (DRAKE_RGBD_CAMERA_IMAGES images), with frames 3:
  Did not find 'camera0' in message
  Did not find 'camera1' in message
  Did not find 'camera2' in message
```

but the documentation in image_array_t explicitly says that frame_name can be empty if each image_t in `images` contains the information in it.

RussTedrake, 2018-11-16:
I was wrong about this.  I've updated the title (@sammy-tri feel free to delegate it to a more appropriate owner).

The problem was because my drake_visualizer instance had latched on to a previous image name, and was not robust to me publishing a new name.

This is a bit painful, because running `bazel test` can publish images, therefore causing drake_visualizer to latch onto them, and then fail to show the published images from my main test/app.  In practice, it means I now have to restart drake_visualizer much more regularly.

jwnimmer-tri, 2021-12-20:
> This is a bit painful, because running bazel test can publish images ...

In general, tests should stay within the sandbox (the `memq://` LCM URL is the `LCM_DEFAULT_URL` during testing) and never publish LCM onto the wire, even if the publication is localhost-only.

In any case, given the pending deprecation in #16215, I wonder it we should close this issue as "won't fix"?

jwnimmer-tri, 2022-05-01:
> In any case, given the pending deprecation in https://github.com/RobotLocomotion/drake/issues/16215, I wonder it we should close this issue as "won't fix"?

Done.

closed 2022-05-01 by jwnimmer-tri