RobotLocomotion/drake #10048: Warm start OSQP solver

hongkai-dai, 2018-11-16:
OSQP supports warm-starting. We haven't used this feature yet, and we should support it.

jwnimmer-tri, 2021-05-05:
FYI Here's the basic mechanism for how `OsqpSolver` can have reuse storage between calls to `Solve`.  It can keep private data within the `MathematicalProgramResult`.

```patch
diff --git a/solvers/osqp_solver.cc b/solvers/osqp_solver.cc
index ed6cc58e1..316cb2b1f 100644
--- a/solvers/osqp_solver.cc
+++ b/solvers/osqp_solver.cc
@@ -270,8 +270,20 @@ void SetDualSolution(
                                    constraint.evaluator()->num_constraints()));
   }
 }
+
+struct PrivateWorkspace {
+  // TODO(jwnimmer-tri) Add OSQP's storage types here, for reuse across calls
+  // to Solve.
+};
+
 }  // namespace
 
+OsqpSolverDetails::OsqpSolverDetails() {
+  private_workspace = AbstractValue::Make(PrivateWorkspace{});
+}
+
+OsqpSolverDetails::~OsqpSolverDetails() = default;
+
 bool OsqpSolver::is_available() { return true; }
 
 void OsqpSolver::DoSolve(
@@ -286,6 +298,11 @@ void OsqpSolver::DoSolve(
 
   OsqpSolverDetails& solver_details =
       result->SetSolverDetailsType<OsqpSolverDetails>();
+  PrivateWorkspace& private_workspace =
+      solver_details.private_workspace->get_mutable_value<PrivateWorkspace>();
+
+  // TODO(jwnimmer-tri) Use me for storing the OSQP workspace for warm-start.
+  unused(private_workspace);
 
   // TODO(hongkai.dai): OSQP uses initial guess to warm start.
   unused(initial_guess);
diff --git a/solvers/osqp_solver.h b/solvers/osqp_solver.h
index 503391dc1..21752a6eb 100644
--- a/solvers/osqp_solver.h
+++ b/solvers/osqp_solver.h
@@ -11,6 +11,11 @@ namespace solvers {
  * details.
  */
 struct OsqpSolverDetails {
+  DRAKE_DEFAULT_COPY_AND_MOVE_AND_ASSIGN(OsqpSolverDetails)
+
+  OsqpSolverDetails();
+  ~OsqpSolverDetails();
+
   /// Number of iterations taken.
   int iter{};
   /// Status of the solver at termination. Please refer to
@@ -33,6 +38,9 @@ struct OsqpSolverDetails {
   /// the problem. Notice that the order of the linear constraints are linear
   /// inequality first, and then linear equality constraints.
   Eigen::VectorXd y{};
+
+  /// For private use of the solver back-end, not for any users.
+  copyable_unique_ptr<AbstractValue> private_workspace;
 };
 
 class OsqpSolver final : public SolverBase {
```

jwnimmer-tri, 2021-07-26:
FYI This is slightly more complicated that I suggested above.  I have a prototype at https://github.com/jwnimmer-tri/drake/commits/osqp-heap, for reference.  The main challenge is that the `SolverBase` is actually deleting the pre-allocated details, instead of reusing it.  That will require a separate, dedicated patch ahead of any OSQP work.

EricCousineau-TRI, 2023-05-29:
I tinkered w/ warm-starting (but not workspace preservation) in #19498.
Notably, `osqp_warm_start_x()` alone seems helpful, but not sufficient, for a one-iteration re-solve, so we could land that as-is.
I ran into the , but found a (possibly janky) workaround.
For preserving / piping back in the dual solution, I ran into #16811 but found a workaround / solution.