RobotLocomotion/drake #10082: GeometryState introspection API has unhelpfully mixed spellings

SeanCurtis-TRI, 2018-11-26:
As mentioned [here](https://reviewable.io/reviews/robotlocomotion/drake/10029#-LRIAGit3vto6MggKUNu), some of the introspection methods reflect their cheap cost with `underscore_names()` and some reflect a higher cost with `CamelCaseNames()`. The distinction is not particularly helpful and it would be simpler if they all had a unified spelling.

SeanCurtis-TRI, 2018-12-06:
Also, as noted in [this comment](https://reviewable.io/reviews/robotlocomotion/drake/10134#-LT3UHTP0bTp7sIPuzyj:-LT3UHTP0bTp7sIPuzyk:b-geypci), consider renaming some of the functions to more explicitly indicate the scope of the count.

SeanCurtis-TRI, 2019-01-03:
From a [conversation in a recent PR](https://reviewable.io/reviews/robotlocomotion/drake/10263#-LVFYf7ZEIQlMGiaD_OR), make sure that the names in the public API match the names in the supporting implementation.

tehbelinda, 2020-01-24:
@SeanCurtis-TRI Does it make sense to change the public API in SceneGraphInspector to match GeometryState and deprecate the older methods? Or should GeometryState be changed to match SceneGraphInspector to reduce public changes? Or a bit of both depending on the method name? E.g. do we want to propagate the use of GetNumXXX vs NumXXX?

SeanCurtis-TRI, 2020-01-24:
I'd propose we come with a principled philosophy for naming and then make all the names match that philosophy. The philosophy should take the various linked conversations into account. So, possibly a bit of both.

Generally, I think I favor replacing `GetNumXXX` with `NumXXX`. Unless I do the triage myself now to refresh my understanding, I can't produce any more helpful feedback.

SeanCurtis-TRI, 2020-05-14:
As of today, `QueryObject` is sane and consistent. There is one apparent contradiction with the `GetRenderEngineByName` -- because there are already three queries `Render*Image` it seemed incongruous to have the word `Render` lead the function as both a verb and a noun.

SceneGraphInspector has the following methods that would benefit from changing:
 - [ ] `all_frame_ids()` This returns a *range* iterator (and deserves the lower case), but the name should reflect range iteration more obviousy.
 - [ ] `GetAllGeometryIds()` This stands in stark contrast to `all_frame_ids()`. This isn't a range but actually collects up all ids in a vector (hence the cost reflected in the name). There should be an (obviously named) range iterator version for this one. And this should lose the `Get`.
 - [ ] `GetCollisionCandidates()` - remove "Get"
 - [ ] `GetSourceName()` - remove "Get"
 - [ ] `GetOwningSourceName()` - remove "Get"
 - [ ] `GetName()` - remove "Get", consider calling it `FrameName` (and `GeometryName` below)
 - [ ] `GetFrameGroup()`
 - [ ] `GetGeometryIdByName()`
 - [ ] `GetFrameId()` - remove "Get", does this need greater implication relationship between the return value (`FrameId`) and parameter value (`GeometryId`)?
 - [ ] `GetShape()`
 - [ ] `GetPoseInParent()`
 - [ ] `GetPoseInFrame()`
 - [ ] `Get * Properties()` - remove "Get". Should there be a "Maybe"? Or "IfAssigned"