RobotLocomotion/drake #10087: Conditional jump or move depends on uninitialized value in MosekTest.TestLogFile

jamiesnape, 2018-11-26:
On Bionic under Valgrind:
```
[ RUN      ] MosekTest.TestLogFile

Conditional jump or move depends on uninitialised value(s)
  at 0xB7E2E97: __wcsnlen_avx2 (strlen-avx2.S:275)
  by 0xB710EC1: wcsrtombs (wcsrtombs.c:104)
  by 0x726A24E: MSK_sysenv_wfopen (in libmosek64.so.8.1)
  by 0x7269FC2: MSK_sysenv_fopen (in libmosek64.so.8.1)
  by 0x707436A: MSK_fopentask (in libmosek64.so.8.1)
  by 0x70974DA: MSK_linkfiletotaskstream (in libmosek64.so.8.1)
  by 0x315DFD: drake::solvers::MosekSolver::Solve(drake::solvers::MathematicalProgram const&, stx::optional<Eigen::Matrix<double, -1, 1, 0, -1, 1> > const&, stx::optional<drake::solvers::SolverOptions> const&, drake::solvers::MathematicalProgramResult*) const (mosek_solver.cc:733)
  by 0x316A11: drake::solvers::MosekSolver::Solve(drake::solvers::MathematicalProgram&) const (mosek_solver.cc:837)
  by 0x13C409: drake::solvers::test::MosekTest_TestLogFile_Test::TestBody() (mosek_solver_test.cc:144)
  by 0x4ED3A3: HandleSehExceptionsInMethodIfSupported<testing::Test, void> (gtest.cc:2402)
  by 0x4ED3A3: void testing::internal::HandleExceptionsInMethodIfSupported<testing::Test, void>(testing::Test*, void (testing::Test::*)(), char const*) (gtest.cc:2438)
  by 0x4DF880: testing::Test::Run() (gtest.cc:2474)
  by 0x4E057F: testing::TestInfo::Run() (gtest.cc:2656)
  by 0x4E0BA6: testing::TestCase::Run() (gtest.cc:2774)
  by 0x4E8D46: testing::internal::UnitTestImpl::RunAllTests() (gtest.cc:4649)
  by 0x4EDF13: HandleSehExceptionsInMethodIfSupported<testing::internal::UnitTestImpl, bool> (gtest.cc:2402)
  by 0x4EDF13: bool testing::internal::HandleExceptionsInMethodIfSupported<testing::internal::UnitTestImpl, bool>(testing::internal::UnitTestImpl*, bool (testing::internal::UnitTestImpl::*)(), char const*) (gtest.cc:2438)
  by 0x4E88FE: testing::UnitTest::Run() (gtest.cc:4257)
Uninitialised value was created by a heap allocation
  at 0x4C2FB0F: malloc (in /usr/lib/valgrind/vgpreload_memcheck-amd64-linux.so)
  by 0x7269E93: defaultmalloc (in libmosek64.so.8.1)
  by 0x726C26E: MSK_memman_malloc (in libmosek64.so.8.1)
  by 0x7269F64: MSK_sysenv_fopen (in libmosek64.so.8.1)
  by 0x707436A: MSK_fopentask (in libmosek64.so.8.1)
  by 0x70974DA: MSK_linkfiletotaskstream (in libmosek64.so.8.1)
  by 0x315DFD: drake::solvers::MosekSolver::Solve(drake::solvers::MathematicalProgram const&, stx::optional<Eigen::Matrix<double, -1, 1, 0, -1, 1> > const&, stx::optional<drake::solvers::SolverOptions> const&, drake::solvers::MathematicalProgramResult*) const (mosek_solver.cc:733)
  by 0x316A11: drake::solvers::MosekSolver::Solve(drake::solvers::MathematicalProgram&) const (mosek_solver.cc:837)
  by 0x13C409: drake::solvers::test::MosekTest_TestLogFile_Test::TestBody() (mosek_solver_test.cc:144)
  by 0x4ED3A3: HandleSehExceptionsInMethodIfSupported<testing::Test, void> (gtest.cc:2402)
  by 0x4ED3A3: void testing::internal::HandleExceptionsInMethodIfSupported<testing::Test, void>(testing::Test*, void (testing::Test::*)(), char const*) (gtest.cc:2438)
  by 0x4DF880: testing::Test::Run() (gtest.cc:2474)
  by 0x4E057F: testing::TestInfo::Run() (gtest.cc:2656)
  by 0x4E0BA6: testing::TestCase::Run() (gtest.cc:2774)
  by 0x4E8D46: testing::internal::UnitTestImpl::RunAllTests() (gtest.cc:4649)
  by 0x4EDF13: HandleSehExceptionsInMethodIfSupported<testing::internal::UnitTestImpl, bool> (gtest.cc:2402)
  by 0x4EDF13: bool testing::internal::HandleExceptionsInMethodIfSupported<testing::internal::UnitTestImpl, bool>(testing::internal::UnitTestImpl*, bool (testing::internal::UnitTestImpl::*)(), char const*) (gtest.cc:2438)
  by 0x4E88FE: testing::UnitTest::Run() (gtest.cc:4257)

Conditional jump or move depends on uninitialised value(s)
  at 0xB679A13: internal_ascii_loop (loop.c:298)
  by 0xB679A13: __gconv_transform_internal_ascii (skeleton.c:609)
  by 0xB710EF4: wcsrtombs (wcsrtombs.c:110)
  by 0x726A24E: MSK_sysenv_wfopen (in libmosek64.so.8.1)
  by 0x7269FC2: MSK_sysenv_fopen (in libmosek64.so.8.1)
  by 0x707436A: MSK_fopentask (in libmosek64.so.8.1)
  by 0x70974DA: MSK_linkfiletotaskstream (in libmosek64.so.8.1)
  by 0x315DFD: drake::solvers::MosekSolver::Solve(drake::solvers::MathematicalProgram const&, stx::optional<Eigen::Matrix<double, -1, 1, 0, -1, 1> > const&, stx::optional<drake::solvers::SolverOptions> const&, drake::solvers::MathematicalProgramResult*) const (mosek_solver.cc:733)
  by 0x316A11: drake::solvers::MosekSolver::Solve(drake::solvers::MathematicalProgram&) const (mosek_solver.cc:837)
  by 0x13C409: drake::solvers::test::MosekTest_TestLogFile_Test::TestBody() (mosek_solver_test.cc:144)
  by 0x4ED3A3: HandleSehExceptionsInMethodIfSupported<testing::Test, void> (gtest.cc:2402)
  by 0x4ED3A3: void testing::internal::HandleExceptionsInMethodIfSupported<testing::Test, void>(testing::Test*, void (testing::Test::*)(), char const*) (gtest.cc:2438)
  by 0x4DF880: testing::Test::Run() (gtest.cc:2474)
  by 0x4E057F: testing::TestInfo::Run() (gtest.cc:2656)
  by 0x4E0BA6: testing::TestCase::Run() (gtest.cc:2774)
  by 0x4E8D46: testing::internal::UnitTestImpl::RunAllTests() (gtest.cc:4649)
  by 0x4EDF13: HandleSehExceptionsInMethodIfSupported<testing::internal::UnitTestImpl, bool> (gtest.cc:2402)
  by 0x4EDF13: bool testing::internal::HandleExceptionsInMethodIfSupported<testing::internal::UnitTestImpl, bool>(testing::internal::UnitTestImpl*, bool (testing::internal::UnitTestImpl::*)(), char const*) (gtest.cc:2438)
  by 0x4E88FE: testing::UnitTest::Run() (gtest.cc:4257)
Uninitialised value was created by a heap allocation
  at 0x4C2FB0F: malloc (in /usr/lib/valgrind/vgpreload_memcheck-amd64-linux.so)
  by 0x7269E93: defaultmalloc (in libmosek64.so.8.1)
  by 0x726C26E: MSK_memman_malloc (in libmosek64.so.8.1)
  by 0x7269F64: MSK_sysenv_fopen (in libmosek64.so.8.1)
  by 0x707436A: MSK_fopentask (in libmosek64.so.8.1)
  by 0x70974DA: MSK_linkfiletotaskstream (in libmosek64.so.8.1)
[... 68 lines cut ...]
  by 0x4E0BA6: testing::TestCase::Run() (gtest.cc:2774)
  by 0x4E8D46: testing::internal::UnitTestImpl::RunAllTests() (gtest.cc:4649)
  by 0x4EDF13: HandleSehExceptionsInMethodIfSupported<testing::internal::UnitTestImpl, bool> (gtest.cc:2402)
  by 0x4EDF13: bool testing::internal::HandleExceptionsInMethodIfSupported<testing::internal::UnitTestImpl, bool>(testing::internal::UnitTestImpl*, bool (testing::internal::UnitTestImpl::*)(), char const*) (gtest.cc:2438)
  by 0x4E88FE: testing::UnitTest::Run() (gtest.cc:4257)
Uninitialised value was created by a heap allocation
  at 0x4C2FB0F: malloc (in /usr/lib/valgrind/vgpreload_memcheck-amd64-linux.so)
  by 0x7269E93: defaultmalloc (in libmosek64.so.8.1)
  by 0x726C26E: MSK_memman_malloc (in libmosek64.so.8.1)
  by 0x7269F64: MSK_sysenv_fopen (in libmosek64.so.8.1)
  by 0x707436A: MSK_fopentask (in libmosek64.so.8.1)
  by 0x70974DA: MSK_linkfiletotaskstream (in libmosek64.so.8.1)
  by 0x315DFD: drake::solvers::MosekSolver::Solve(drake::solvers::MathematicalProgram const&, stx::optional<Eigen::Matrix<double, -1, 1, 0, -1, 1> > const&, stx::optional<drake::solvers::SolverOptions> const&, drake::solvers::MathematicalProgramResult*) const (mosek_solver.cc:733)
  by 0x316A11: drake::solvers::MosekSolver::Solve(drake::solvers::MathematicalProgram&) const (mosek_solver.cc:837)
  by 0x13C409: drake::solvers::test::MosekTest_TestLogFile_Test::TestBody() (mosek_solver_test.cc:144)
  by 0x4ED3A3: HandleSehExceptionsInMethodIfSupported<testing::Test, void> (gtest.cc:2402)
  by 0x4ED3A3: void testing::internal::HandleExceptionsInMethodIfSupported<testing::Test, void>(testing::Test*, void (testing::Test::*)(), char const*) (gtest.cc:2438)
  by 0x4DF880: testing::Test::Run() (gtest.cc:2474)
  by 0x4E057F: testing::TestInfo::Run() (gtest.cc:2656)
  by 0x4E0BA6: testing::TestCase::Run() (gtest.cc:2774)
  by 0x4E8D46: testing::internal::UnitTestImpl::RunAllTests() (gtest.cc:4649)
  by 0x4EDF13: HandleSehExceptionsInMethodIfSupported<testing::internal::UnitTestImpl, bool> (gtest.cc:2402)
  by 0x4EDF13: bool testing::internal::HandleExceptionsInMethodIfSupported<testing::internal::UnitTestImpl, bool>(testing::internal::UnitTestImpl*, bool (testing::internal::UnitTestImpl::*)(), char const*) (gtest.cc:2438)
  by 0x4E88FE: testing::UnitTest::Run() (gtest.cc:4257)
```

jamiesnape, 2018-11-27:
My initial thoughts are either this a false-positive relating to the implementation of `wcsrtombs` or a defect with MOSEK's use of `wcsrtombs`. Seems less much likely to be a problem within Drake.

jamiesnape, 2018-12-03:
Leaving this open for @hongkai-dai to confirm that the suppression is reasonable when he returns.

jwnimmer-tri, 2020-04-21:
@hongkai-dai Please review the suppression added in #10095 and then close out this issue.

hongkai-dai, 2020-04-21:
Thanks for fixing the issue. Sorry for my belated response.

closed 2020-04-21 by hongkai-dai