RobotLocomotion/drake #10098: Need sugar for DeclarePeriodicEvent to take handler functions

sherm1, 2018-11-28:
The preferred way to define events in Drake is to provide a handler function rather than to override one of the dispatch methods like DoPublish(). Currently (see PR #9766 examples/fibonacci) that leads to boilerplate clutter that could be removed with some sugar like we have provided elsewhere. We currently have: 
```c++
    // Output yₙ using a Drake "publish" event (occurs at the end of step n).
    DeclarePeriodicEvent(kPeriod, 0.,  // period, offset
                         systems::PublishEvent<double>(
                             [this](const systems::Context<double>& context_n,
                                    const systems::PublishEvent<double>&) {
                               Output(context_n);
                             }));
```
which should look more like this:
```c++
    // Output yₙ using a Drake "publish" event (occurs at the end of step n).
    DeclarePeriodicPublishEvent(kPeriod, 0., &SystemName::Output);
```
Discrete and Unrestricted updates should get similar sugar, with only the form of the handler function changing.

SeanCurtis-TRI, 2018-11-28:
[ImageWriter](https://github.com/RobotLocomotion/drake/blob/master/systems/sensors/image_writer.cc#L149) had a similar issue and would likewise benefit from this sugar.

sherm1, 2019-01-05:
Fixed by #10321

closed 2019-01-05 by sherm1