RobotLocomotion/drake #10123: Binary packages for Bionic contain prerequisite setup scripts for Xenial

jamiesnape, 2018-11-30:
This assumption is invalid:

https://github.com/RobotLocomotion/drake/blob/4c886e1d29c63c0bb84ec65f9f0c4fb7d163f02e/setup/BUILD.bazel#L17-L20

Relates #7882.

jamiesnape, 2018-11-30:
FYI @EricCousineau-TRI.

KarimHassanieh, 2018-12-03:
I faced this issue with the setup script...will it be fixed ? Also I wanted to ask if Drake will eventually be integrated with ROS?

EricCousineau-TRI, 2018-12-03:
For resolving this, can we just install all of the prereq scripts, and include a platform-detecting dispatch shell script?

This seems simpler than making an `@os` repository (which may be useful at some point in the future, but may be more than necessary for now) or trying to make some other sort of configuration flag that can be used. We could then strip out the `select` statements.

EricCousineau-TRI, 2018-12-03:
Regarding ROS, we do have plans for providing integration components; will post an issue and cc you in it.

jamiesnape, 2018-12-03:
Certainly there is enough commonality in the scripts that there should probably only be one script and two package lists, but in the interim, please grab the scripts directly from the repo:

```
wget https://raw.githubusercontent.com/RobotLocomotion/drake/master/setup/ubuntu/18.04/binary_distribution/install_prereqs.sh
wget https://raw.githubusercontent.com/RobotLocomotion/drake/master/setup/ubuntu/18.04/binary_distribution/packages.txt
chmod +x install_prereqs.sh
sudo ./install_prereqs.sh
```

jwnimmer-tri, 2018-12-03:
FYI To whoever works on the PR for this, here's the snippets we use in Anzu to support Xenial and Bionic for install_prereqs.  You are welcome to copy them into Drake if they are useful.

```sh

cat_without_comments() {
    sed -e 's|#.*||g;' "$(dirname $0)/${ubuntu_release_yymm}/$1"
}

# Run `apt-get install` on the arguments, then clean up.
apt_install () {
  apt-get -q install --no-install-recommends "$@"
  apt-get -q clean
}

ubuntu_release_yymm=$(lsb_release -sr)
[[ "${ubuntu_release_yymm}" == "16.04" ]] ||
[[ "${ubuntu_release_yymm}" == "18.04" ]] ||
die "${me} only supports Ubuntu 16.04 or 18.04."

apt_install $(cat_without_comments packages-python.txt)
apt_install $(cat_without_comments packages-ubuntu.txt)
```

jamiesnape, 2018-12-03:
I think its on my plate. Probably just a rename of package.txt files to be distro specific and then a substitution in the script.

PR #10141 (2018-12-07): Consolidate Ubuntu prerequisite install scripts and cleanup associated Dockerfiles
Fixes #10123. Loosely relates  #7882.

----
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/10141)
closed 2018-12-07 by EricCousineau-TRI