RobotLocomotion/drake #10137: cryptic error message from scene_graph_parser_detail when user passes in an unsupported mesh

RussTedrake, 2018-12-04:
From a student:

I tried to build my simulation using a mesh file as follow:
```
<geometry>
  <mesh>
    <uri>file:///C:/Users/Quang Le/Dropbox/MIT/Course 6/6.881/Pset/lab2_code/lab2_opening_door/models/hole.dae</uri>
  </mesh>
</geometry>
```
 And I got this error:
```
RuntimeError: multibody/multibody_tree/parsing/scene_graph_parser_detail.cc: ResolveVisualUri: ERROR: Mesh file name could not be resolved from the provided uri "file:///C:/Users/Quang Le/Dropbox/MIT/Course 6/6.881/Pset/lab2_code/lab2_opening_door/models/hole.dae".
```

jamiesnape, 2018-12-04:
Does not seem cryptic to me. It is a URI and it cannot be resolved. There is a space in the URI, which makes it not invalid, regardless of whether the target exists. Also it is on a Windows filesystem, which is not tested by us.

SeanCurtis-TRI, 2018-12-04:
Looking in detail at this, I tend to agree with @jamiesnape. This clearly isn't an issue of "obj" vs "dae". This is a case of "you've provided a URI that doesn't indicate an actual file I can locate." Admittedly, the error message could be more direct. But not by much.

RussTedrake, 2018-12-04:
You’re right.  I didn’t inspect it closely enough.