RobotLocomotion/drake #10139: Problem with VTK starting drake_visualizer (Ubuntu 16.04)

hwiedDFKI, 2018-12-04:
## Description

I try to run examples/humanoid_controller as described in it's README. Unfortunately starting the drake_visualizer by ```./bazel-bin/tools/drake_visualizer``` leads to the follwing error:
```
~/git/drake (master)$ ./bazel-bin/tools/drake_visualizer
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/home/hwied/.cache/bazel/_bazel_hwied/95523a503ac442241d49857d991af123/external/drake_visualizer/lib/python2.7/site-packages/director/drakevisualizerapp.py", line 3, in <module>
    from director import mainwindowapp
  File "/home/hwied/.cache/bazel/_bazel_hwied/95523a503ac442241d49857d991af123/external/drake_visualizer/lib/python2.7/site-packages/director/mainwindowapp.py", line 2, in <module>
    from director.componentgraph import ComponentFactory
  File "/home/hwied/.cache/bazel/_bazel_hwied/95523a503ac442241d49857d991af123/external/drake_visualizer/lib/python2.7/site-packages/director/componentgraph.py", line 2, in <module>
    from director.fieldcontainer import FieldContainer
  File "/home/hwied/.cache/bazel/_bazel_hwied/95523a503ac442241d49857d991af123/external/drake_visualizer/lib/python2.7/site-packages/director/fieldcontainer.py", line 1, in <module>
    import vtk
  File "/usr/lib/python2.7/dist-packages/vtk/__init__.py", line 39, in <module>
    from vtkCommonCore import *
  File "/usr/lib/python2.7/dist-packages/vtk/vtkCommonCore.py", line 1, in <module>
    from vtkCommonCorePython import *
TypeError: value must be a class or type
Segmentation fault (core dumped)
```
```bazel run //tools:drake_visualizer``` results in the same error.

Thanks for your help in advance. :)

## Debug Information

I'm on the master branch at commit 365e7f9d. Build with Bazel under Ubuntu 16.04 using python2.7:
```
+ which python2
/usr/bin/python2
+ python2 --version
Python 2.7.12
```

vtk:

```
~/git/drake (master)$ apt list --installed | grep vtk

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

libvtk6-dev/xenial,now 6.2.0+dfsg1-10build1+debian11.1+osrf1 amd64 [installed,automatic]
libvtk6-java/xenial,now 6.2.0+dfsg1-10build1+debian11.1+osrf1 amd64 [installed,automatic]
libvtk6-qt-dev/xenial,now 6.2.0+dfsg1-10build1+debian11.1+osrf1 amd64 [installed,automatic]
libvtk6.2/xenial,now 6.2.0+dfsg1-10build1+debian11.1+osrf1 amd64 [installed,automatic]
libvtk6.2-qt/xenial,now 6.2.0+dfsg1-10build1+debian11.1+osrf1 amd64 [installed,automatic]
python-vtk6/xenial,now 6.2.0+dfsg1-10build1+debian11.1+osrf1 amd64 [installed,automatic]
tcl-vtk6/xenial,now 6.2.0+dfsg1-10build1+debian11.1+osrf1 amd64 [installed,automatic]
vtk6/xenial,now 6.2.0+dfsg1-10build1+debian11.1+osrf1 amd64 [installed,automatic]
~/git/drake (master)$ python -m pip install --upgrade vtk
Collecting vtk
  Using cached https://files.pythonhosted.org/packages/bc/39/99d717feb6246e875188e60cc7200e60f862a457972e32c250a0cb8d2b3d/vtk-8.1.2-cp27-cp27mu-manylinux1_x86_64.whl
Installing collected packages: vtk
Successfully installed vtk-8.1.2
```

bazel & cmake:
```
+ which bazel
/usr/bin/bazel
+ bazel version
Build label: 0.19.2
Build target: bazel-out/k8-opt/bin/src/main/java/com/google/devtools/build/lib/bazel/BazelServer_deploy.jar
Build time: Mon Nov 19 16:25:09 2018 (1542644709)
Build timestamp: 1542644709
Build timestamp as int: 1542644709
+ which cmake
/usr/bin/cmake
+ cmake --version
cmake version 3.5.1

CMake suite maintained and supported by Kitware (kitware.com/cmake).
```
host settings in my present project workspace:
```
~/git/drake (master)$ bazel run @drake//common:print_host_settings
INFO: Analysed target @drake//common:print_host_settings (3 packages loaded, 8 targets configured).
INFO: Found 1 target...
Target @drake//tools/cc_toolchain:print_host_settings up-to-date:
  bazel-bin/external/drake/tools/cc_toolchain/print_host_settings
INFO: Elapsed time: 0.415s, Critical Path: 0.06s, Remote (0.00% of the time): [queue: 0.00%, setup: 0.00%, process: 0.00%]
INFO: 1 process: 1 linux-sandbox.
INFO: Build completed successfully, 5 total actions
INFO: Running command line: bazel-bin/external/drake/tools/cc_toolchain/print_host_settings external/drake/tools/ccINFO: Build completed successfully, 5 total actions
+ capture_cc_env=external/drake/tools/cc_toolchain/capture_cc.env
+ source external/drake/tools/cc_toolchain/capture_cc.env
++ BAZEL_CC=/usr/bin/gcc
+ /usr/bin/gcc --version
gcc (Ubuntu 5.4.0-6ubuntu1~16.04.10) 5.4.0 20160609
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

```

compilers being used in my  project:
```
~/git/drake (master)$ cmake -LA . | grep 'CMAKE_.*_COMPILER'
CMAKE_CXX_COMPILER:FILEPATH=/usr/bin/g++
CMAKE_C_COMPILER:FILEPATH=/usr/bin/gcc
~/git/drake (master)$ g++ --version
g++ (Ubuntu 5.4.0-6ubuntu1~16.04.10) 5.4.0 20160609
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

~/git/drake (master)$ gcc --version
gcc (Ubuntu 5.4.0-6ubuntu1~16.04.10) 5.4.0 20160609
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

```

jamiesnape, 2018-12-04:
We do not support an apt installed VTK. You must use the VTK bundled with Drake.

jamiesnape, 2018-12-04:
(Try uninstalling both the apt and pip VTK as a first step.)

EricCousineau-TRI, 2018-12-11:
Howdy @hwiedDFKI! This might be failing because there might be additional missing dependencies; can you post the _full_ output, including any lines after the failure?

In addition, can you double-check that `install_prereqs.sh` did not somehow die with an error?