RobotLocomotion/drake #10149: Unrestricted updates slow simulation significantly

sammy-tri, 2018-12-05:
## Description

Adding unrestricted updates to a system significantly slows the simulation.  This appears to be due to the copying of the entire state for the context in `System::CalcUnrestrictedUpdate` (which may be called recursively through diagrams) and again at completion of the update in `Simulator::HandleUnrestrictedUpdate`.  Commenting these lines out eliminates the slowdown (though of course the updates have no longer have any effect).

## Reproduction

`examples/manipulation_station/mock_station_simulation --target_realtime_rate 0` runs at approximately 2x realtime on my workstation when not receiving LCM messages.  Running `
examples/manipulation_station/joint_teleop --hardware` slows the simulation down to about 0.4x due to the unrestricted updates for the incoming LCM messages.

An alternate way to reproduce the issue is to add a no-op unrestricted update somewhere else in the simulation.  For example, `IiwaCommandReceiver` already has a 200Hz discrete update.  Adding a 200Hz unrestricted update in the constructor (with no handling code) slows the simulation from 2.0x to about 0.8x. 

## Debug Information

* What operating system are you running Drake on?
(e.g. Ubuntu 16.04, macOS 10.13)

Ubuntu 16.04

* Is this an issue with (a) C++, (b) Python, or (c) the distributed tools
(e.g. `drake-visualizer`)?

C++

* Are you (a) building `drake` from source, or (b) from a binary release?

From source

* What version of `python` are you using?
```
+ which python2
/usr/bin/python2
+ python2 --version
Python 2.7.12
```
 
* If you are building from source or using the `drake` C++ library as an
external:

    * What version of Bazel and CMake are you using?
```
+ which bazel
/usr/bin/bazel
+ bazel version
WARNING: --batch mode is deprecated. Please instead explicitly shut down your Bazel server using the command "bazel shutdown".
Build label: 0.19.2
Build target: bazel-out/k8-opt/bin/src/main/java/com/google/devtools/build/lib/bazel/BazelServer_deploy.jar
Build time: Mon Nov 19 16:25:09 2018 (1542644709)
Build timestamp: 1542644709
Build timestamp as int: 1542644709
+ which cmake
/usr/bin/cmake
+ cmake --version
cmake version 3.5.1

CMake suite maintained and supported by Kitware (kitware.com/cmake).
```

    * If you are using Bazel, please provide the host settings in your present
    project workspace:
```
+ capture_cc_env=external/drake/tools/cc_toolchain/capture_cc.env
+ source external/drake/tools/cc_toolchain/capture_cc.env
++ BAZEL_CC=/usr/bin/gcc
+ /usr/bin/gcc --version
gcc (Ubuntu 5.4.0-6ubuntu1~16.04.10) 5.4.0 20160609
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

* If you are building `drake`
[from source](https://drake.mit.edu/from_source.html):

    * What git commit of `drake` are you using?
fb00f1d

    * Are you building `drake` with Bazel or CMake?

bazel

    * Are you using `drake` directly, or consuming it as an external?

directly

sherm1, 2018-12-05:
Oh yeah -- I remember discussing this when we first designed this thing! It was a hypothetical problem then ... now it's real. We discussed a number of solutions at the time:
- providing copy-on-write for State so that only things that change actually get copied,
- creating a DeltaState object that would just contain changes, or
- a form of event handler that takes a mutable Context that can be changed directly.

Other ideas?

jwnimmer-tri, 2018-12-05:
Unborking the part-whole state update implementation in System, so that we're only copying _one System's_ diff back and forth for a Diagram, is likely easy and don't involve COW complexities.

jwnimmer-tri, 2018-12-05:
... and then for systems with expensive-to-copy abstract state, given the new declare-callback update mechanisms (instead of virtual overrides), it would be easy to provide an opt-in "alias my output pointer from within the const-Context" which is dangerous in general but avoids extra copies when the function body is aware of the risk.

edrumwri, 2018-12-06:
Your proposal would seem to introduce a lot of implementation complexity from the updating (not clear that COW would be worse in this regard). Also @sherm1 observed that the running time improvement is going to be predicated on the state data being well distributed among the systems.

sherm1, 2018-12-06:
Yes, I think you're right @sammy-tri :(

sammy-tri, 2018-12-07:
I think sherm's approach above is probably good.  My only thought is if we go that route we should either remove the current unrestricted update or at least plaster warnings around it indicating that this can be a major performance hit during simulation.

jwnimmer-tri, 2018-12-07:
I think even without a new event time, it's within reason to allow Diagram to assist it only copying State for the subset of systems who are processing a (batch of) events.  As I said above, "'t would be easy to provide an opt-in "alias my output pointer from within the const-Context" which is dangerous in general but avoids extra copies when the function body is aware of the risk.' and I think Diagram could use that to solve this, without inventing new science.

jwnimmer-tri, 2018-12-07:
In other words, the fundamental defect here is that an event declared on a tiny system buried within a diagram (or sub-sub-diagram) causes the _entire_ simulation's state to be copied.  The mutated scope bloating from the part into the whole is the real issue, so it's really a _Diagram_ problem, not a Simulator or Event problem.

jwnimmer-tri, 2018-12-07:
> ... it would be easy to provide an opt-in "alias my output pointer from within the const-Context" ...

To elaborate on what I mean by this:

The event handler is passed two(ish) things -- the full `const Context&` and a `State* new_state` where it writes the t+ state.  By default, writing into `new_state` should _not_ change the state values seen in `const Context&` because that's surprising / confusing.  However, on an opt-in basis, we could allow handlers to declare that `new_state` should be a mutable pointer aliasing into the same Context, which would allow the code calling the handler to not do any copying.

sherm1, 2018-12-07:
If I understand the proposal, the opt-in idea seems complicated to me and would cause identical-looking handlers to have different ordering behavior. Consider a subsystem like SceneGraph which may have several abstract states, say a gigantic GeometryState and a trivial bool. With the COW approach a handler that modifies only the bool obeys the well-behaved ordering we have now. If SG has to say "alias my State" in order to avoid copying GeometryState (which handlers normally wouldn't be updating), then ordering gets mangled for the simple bool-setting handlers also.

jwnimmer-tri, 2018-12-07:
You're mixing two issues.

You're talking about making a _single_ event handler more performant.  Yes, for that we could use COW, though we could also instead declare that handlers only operate on a subset of state (such as a group number, or ticket), which also could help with dependency analysis.

I'm talking about whether a tiny 100-byte unrestricted update in a leaf system should cause Diagram to lose its mind and clone all of the simulator state twice.  Whether we use the mutable-state-alias for that (in Diagram implementation) or not is beside the point.  The only goal is to fix _Diagram_ to not require copying all of the state twice.

jwnimmer-tri, 2018-12-07:
COW means you're hitting the heap in your simulation loop.  The other kinds of solutions allow for 100% pre-allocation of what's needed.

jwnimmer-tri, 2018-12-07:
In any case, I've completely failed in last year's promise to myself to remain completely uninvolved in the framework's event implementation.  I saw Sam and Siyuan struggling here and wanted to help, but I think that making any progress here is really difficult.  I'm going to step aside from this discussion now.

sherm1, 2018-12-07:
OK, thanks Jeremy.

edrumwri, 2018-12-07:
To reiterate what I said to @sherm1 privately: I think his UnorderedContextUpdate() proposal would solve the problem faster, but I don't like that it gives the user more opportunities to screw things up. I'm still most in favor for the COW proposal since it would not require API changes and would be an elegant solution to the problem. The complexity of mitigating the unrestricted update copying problem would still hard be (but we knew it was hard when we identified it nearly two years ago!)

Jeremy broaches a nice point with the heap allocation, but I expect the heap allocation requests to be typically small (which a smart OS _should_ be able to exploit). Large allocations imply large copies anyway, so I would expect low benefit from preallocation.

sherm1, 2019-04-02:
Update: #11113 addresses the post-update "copy back" part of this issue by copying only State for subsystems that actually handled events. Next I will PR code to do the same for the initial State copy. That should be sufficient to solve this problem for now -- it still copies more than necessary but only within a subsystem that actually handled an event. State for non-event-handling subsystems (like SceneGraph) won't get copied.

closed 2019-04-03 by sherm1

jwnimmer-tri, 2019-04-04:
I repeated the benchmark from the first post.

Condition (1):
  - `examples/manipulation_station/mock_station_simulation --target_realtime_rate 0 &`

Condition (2):
  - `examples/manipulation_station/mock_station_simulation --target_realtime_rate 0 &`
  - `examples/manipulation_station/joint_teleop --hardware &`

As of cf72b348e8a9bb9709286672904fc9512cf1036d on master, I see rates:
(1) 1.45x
(2) 1.15x

With #11158 added (removes the LCM discrete state hack, so that the Diagram is similar to the original report), I see rates:
(1) 1.45x
(2) 1.15x

With #11158 added and 1a98f0a9b095c9808206ab4ef681926bfc736a46 (#11125) reverted, I see rates:
(1) 1.40x
(2) 0.60x

With #11158 applied and 1a98f0a9b095c9808206ab4ef681926bfc736a46 (#11125) and d0dd5b0b9ab35d4a7d08f71ed7e56643c1808401 (#11113) reverted, I see rates:
(1) 1.30x
(2) 0.35x

So yes, with the specific kinds of states and rates as reported in the original post, this does appear to be resolved.

siyuanfeng-tri, 2019-04-04:
here is the next question: why do we have the 1.45 -> 1.15?

sherm1, 2019-04-04:
Yay! Thanks for running & reporting these tests, Jeremy.

PR #11125 (2019-04-03): Selectively copy only the state for event-handling leaf subsystems
Currently `System::CalcDiscreteVariableUpdates()` and `CalcUnrestrictedUpdate()`, when invoked on a Diagram, copy the entire Diagram discrete state or (worse) entire state (resp.) when _any_ event occurs _anywhere_ in a subsystem contained in that Diagram.

This PR delays copying until we recurse down to a leaf subsystem that is actually known to have at least one event that needs handling. Then the full discrete state, or full state (resp.) for that leaf subsystem is copied prior to invoking the handlers. This avoids copying n [... cut]