RobotLocomotion/drake #10153: scene_graph, collision checking: Support for capsules

EricCousineau-TRI, 2018-12-06:
A team at TRI would like to have support for collision checking using capsules.

From briefly looking at the implementation, it looks like `fcl::GEOM_CAPSULE` is not yet supported.

@DamrongGuoy Is it OK if I assign this one to you?

\cc @sherm1

DamrongGuoy, 2018-12-06:
To be sure, we are talking about `drake::geometry::QueryObject<T>::ComputePointPairPenetration()`, right?   We do not need `ComputeSignedDistancePairwiseClosestPoints()` and `ComputeSignedDistanceToPoint()` for #10153, right?

sherm1, 2018-12-06:
My impression from talking with @EricCousineau-TRI is that the group he's referring to uses capsules as a generic shape the way the Anzu folks have been using spheres. The robot arms would be covered in capsules that would be used as the geometry for trajectory planning, so would need distances.

EricCousineau-TRI, 2018-12-19:
I believe the main use cases are:
- `ComputeSignedDistancePairwiseClosestPoints()` for control
- Checking if something is in collision (just binary it seems)

sherm1, 2018-12-19:
(Forwarding from Eric's Slack post for posterity)
Following up on #10153:
Just perused the MMT code; here's use case for pointwise signed distances (QP constraints):
https://github.awsinternal.tri.global/mmt/mobile_manipulation/blob/5581f9d41ee2bf6cdcdc8b23f921906ce1e74461/common/control/qp_inverse_kinematics.cc#L978

Point-in-collision query for voxel-based collision checks (I think):
https://github.awsinternal.tri.global/mmt/mobile_manipulation/blob/5581f9d41ee2bf6cdcdc8b23f921906ce1e74461/modules/voxel_mapper/voxel_mapper_module.cc#L551

Minimum distance (seems like its for subtracting the robot from a mask):
https://github.awsinternal.tri.global/mmt/mobile_manipulation/blob/5581f9d41ee2bf6cdcdc8b23f921906ce1e74461/rippers/mask_robot/mask_robot_ripper_module.cc#L70-L71

sherm1, 2019-10-11:
We also want capsules to work for point contact, and as rigid or soft objects in hydroelastics (meaning we need tri and tet meshes and an extent field).

tehbelinda, 2019-11-01:
As per Sherm's comment, capsule can be implemented using the Apache licensed vtkCapsuleSource https://github.com/finetjul/bender/tree/master/Libs/VTK/Filters/Sources

Notes for adding it to our repo from Jeremy:

> 
> On first look, yes it would be OK to copy the h and cxx into drake/third_party/com_github_com_finetjul_bender/... and maintain it there.
> 
> We'd need to document the upstream location and git sha we've copied.
> 
> We'd need to copy (or extract) a LICENSE.txt file for those files and add that license file to the install rules so we provide the notice required by the upstream copyright.
> 
> We'd need to be sure not to install the header, and to compile the cc_library with -fvisibility=hidden to keep the symbols private.

tehbelinda, 2019-12-07:
Closing this out now that these functions are supported - will make a separate issue specifically for `ComputeSignedDistancePairwiseClosestPoints` since it has varying amounts of support for different shapes and scalar types

closed 2019-12-07 by tehbelinda