RobotLocomotion/drake #10155: Rename MBT Jacobian Methods

avalenzu, 2018-12-06:
After much f2f discussion, @amcastro-tri, @avalenzu, @edrumwri, @mitiguy, @RussTedrake, and @sherm1, have converged on the following nomenclature for Jacobian methods in `MultibodyTree`.

Enumerated type:
```c++
enum class JacobianWrtVariable { kWrtQDot,  kWrtV };
```

General method names: 
1. `CalcJacobianSpatialVelocity(	context, kWrtVariable, frameB, p_BP, frameA, expressedInFrame, ...)`
2. `CalcJacobianAngularVelocity(	context, kWrtVariable, frameB, frameA, expressedInFrame, ...)`
3. `CalcJacobianTranslationalVelocity(	context, kWrtVariable, frameB, p_BP, frameA, expressedInFrame, ...)`
Note: `p_BP` is the position vector from Bo (frame B’s origin) to a point P (regarded as a point of B).

#10122 adds `CalcJacobianSpatialVelocity()`, but does not rename existing methods. The remaining methods should be renamed and the old names deprecated.

Related to #9693, #9971.

EricCousineau-TRI, 2018-12-10:
Just a curiosity for short names: is there a way that we can strip `Variable` off the end, since it can be implicit, so we get the shortened `JacobianWrt::kQDot`, `JacobianWrt::kV`?

avalenzu, 2018-12-10:
We considered this, but felt that it makes the declaration harder to read:
```c++
CalcJacobianSpatialVelocity(..., JacobianWrt with_respect_to, ...)
```
In code where it's going to be used a lot, you can always do:
```
using JacobianWrt = JacobianWrtVariable ;
```

tri-ltyyu, 2019-01-25:
@mitiguy , please reference PR to this

tri-ltyyu, 2019-03-29:
Will re-visit this after block example is completed.

amcastro-tri, 2019-04-29:
Any updates on this @mitiguy? should we re-assign? 
A first spiral would only entail a refactor according to our previous discussions, deprecation, and pretty much no doc changes at all. This first spiral could land as soon as this week.
A second spiral could entail a more in depth re-work of the docs per @mitiguy's research on new notation.

IMO the second spiral should not block the first spiral.

amcastro-tri, 2019-05-13:
I believe @mitiguy is working on the first spiral which simply entails:
- Refactor methods to the names we discussed in this thread.
- Reuse (copy/paste) docs from the already available docs.
- Deprecated older method names.

@mitiguy please provide any updated as you consider appropriate.

mitiguy, 2019-05-14:
Restarting these changes now (May 14 2019).
If all goes well, I should be finished with first spiral soon.

mitiguy, 2019-05-15:
Mitiguy's working notes for subsequent PRs on this issue.
The general method names (detailed above) already exist.
The following Jacobian methods exist and were marked by @amcastro-tri for deprecation.

 void CalcPointsGeometricJacobianExpressedInWorld(
      const systems::Context<T>& context,
      const Frame<T>& frame_F, const Eigen::Ref<const MatrixX<T>>& p_FP_list,
      EigenPtr<MatrixX<T>> p_WP_list, EigenPtr<MatrixX<T>> Jv_WFp) 
This method maps mostly to the new general method CalcJacobianTranslationalVelocity( ...)
However, it also returns extra information in p_WP_list.

  void CalcPointsGeometricJacobianExpressedInWorld(
      const systems::Context<T>& context,
      const Frame<T>& frame_F, const Eigen::Ref<const MatrixX<T>>& p_WP_list,
      EigenPtr<MatrixX<T>> Jv_WFp) 
This method maps mostly to the new general method CalcJacobianTranslationalVelocity( ...).
However, its input argument p_WP_list differs from p_FP_list.

  void CalcPointsAnalyticalJacobianExpressedInWorld(
      const systems::Context<T>& context,
      const Frame<T>& frame_F, const Eigen::Ref<const MatrixX<T>>& p_FP_list,
      EigenPtr<MatrixX<T>> p_WP_list, EigenPtr<MatrixX<T>> Jq_WFp) 
This method maps mostly to the new general method CalcJacobianTranslationalVelocity( ...)
However, it also returns extra information in p_WP_list.  

  void CalcFrameGeometricJacobianExpressedInWorld(
      const systems::Context<T>& context,
      const Frame<T>& frame_F, const Eigen::Ref<const Vector3<T>>& p_FP,
      EigenPtr<MatrixX<T>> Jv_WFp) 
This method maps mostly to the new general method CalcJacobianSpatialVelocity(...)
 
 void CalcRelativeFrameGeometricJacobian(
      const systems::Context<T>& context,
      const Frame<T>& frame_B, const Eigen::Ref<const Vector3<T>>& p_BP,
      const Frame<T>& frame_A, const Frame<T>& frame_E,
      EigenPtr<MatrixX<T>> Jv_ABp_E) 
This method maps mostly to the new general method CalcJacobianSpatialVelocity(...)

VectorX<T> CalcBiasForPointsGeometricJacobianExpressedInWorld( ...)
This method does not map to any new general Jacobian method.
However, it needs to be renamed for compatibility with the newer Jacobian methods, e.g.,
some variant of CalcBiasForJacobianTranslationalVelocity...(....)

  Vector6<T> CalcBiasForFrameGeometricJacobianExpressedInWorld(
      const systems::Context<T>& context,
      const Frame<T>& frame_F, const Eigen::Ref<const Vector3<T>>& p_FP) 
This method does not map to any new general Jacobian method.
However, it needs to be renamed for compatibility with the newer Jacobian methods, e.g.,
some variant of CalcBiasForJacobianSpatialVelocity...(....)

amcastro-tri, 2019-05-15:
This is difficult to parse and I believe a bulleted list of about ~6 items would suffice. What about creating a buleted list with the format:
 - OldApi1Name() --> NewApi1Name()
 - OldApi2Name() --> NewApi2Name()
 - etc.

The arrow implicitly says that: 1) OldApiXName() will get deprecated. 2) NewApiXName() is the new replacement according to our design doc.

mitiguy, 2019-05-20:
Progress update Monday May 20th -- background information only
Merged PR#11471 last week to complete the following checkbox in this issue:
Deprecate CalcRelativeFrameGeometricJacobian(...).  in favor of CalcJacobianSpatialVelocity(...)

Created PR#11495 which is currently under review to address next checkbox in this issue:
Rename CalcBiasForFrameGeometricJacobianExpressedInWorld(...) to
CalcBiasForJacobianSpatialVelocity(...)

mitiguy, 2019-06-12:
Merged PR #11495 last week (June 3, 2019) to complete the following checkbox in this issue:
Rename CalcBiasForFrameGeometricJacobianExpressedInWorld(...) to
CalcBiasForJacobianSpatialVelocity(...)

Created PR #11623 (June 11, 2019) which is under review to address next two checkboxes in this issue:
Deprecate two variants of CalcPointsGeometricJacobianExpressedInWorld(...)
in favor of CalcJacobianTranslationalVelocity( ...).

mitiguy, 2019-06-18:
Merged PR #11623 June 17, 2019 to complete the following checkbox in this issue:
Deprecate two variants of CalcPointsGeometricJacobianExpressedInWorld(...)
in favor of CalcJacobianTranslationalVelocity( ...).

Created PR #11xxx  (June 18, 2019) to address next checkbox in this issue:
Deprecate CalcPointsAnalyticalJacobianExpressedInWorld(...)
in favor of CalcJacobianTranslationalVelocity( ...).

mitiguy, 2019-06-24:
Merged PR #11671 June 24, 2019 to complete the following checkbox in this issue:
Deprecate CalcPointsAnalyticalJacobianExpressedInWorld(...)
in favor of CalcJacobianTranslationalVelocity( ...).

Soon to create PR #xxxxx (June yy, 2019) to address final checkbox in this issue:

sherm1, 2020-05-12:
@mitiguy can this be closed now?

jwnimmer-tri, 2021-11-12:
@mitiguy what work still remains before we can close this?

jwnimmer-tri, 2023-05-23:
Closing for lack of response.

closed 2023-05-23 by jwnimmer-tri