RobotLocomotion/drake #10159: Undefined function 'getTerrainContactPoints' in Atlas example

Jerryxiaoyu, 2018-12-07:
Hello guys, I reinstall the old version of `last_sha_with_original_matlab`.  
But when I tried to execute` runAtlasWalking.m` on Ubuntu 16.04 Matlab2017b,  error occurred as follows:
```
ans =

  logical

   1

Undefined function 'getTerrainContactPoints' for input arguments of type 'Atlas'.

Error in RigidBodySupportState (line 41)
          terrain_contact_point_struct = getTerrainContactPoints(r,obj.bodies(i));

Error in Atlas (line 114)
      obj.left_full_support = RigidBodySupportState(obj,obj.foot_body_id.left);

Error in runAtlasWalking (line 26)
r = Atlas(fullfile(getDrakePath,'examples','Atlas','urdf','atlas_minimal_contact.urdf'),robot_options);

Error in runAtlasFastWalking (line 3)
runAtlasWalking([], struct('navgoal', [2.0;0;0;0;0;0],...
```
 
It seems to be related with matlab license #3777 #4250. But I don't know how to fix it. 
Actually, before I re-installed system, I have successfully run these examples and everything was okay. So I don't know if there is something with matlab config or Cmake now.
Could anyone help me? thank you.

EricCousineau-TRI, 2018-12-10:
Howdy @Jerryxiaoyu! Unfortunately, this is no longer a supported configuration that we're actively developing on, so unfortunately I'll be closing this issue.

However, I briefly tried it out on my side. Due to licensing constraints, I cannot execute `resolveConstraints` in the Atlas constructor at the moment.
That being said, I was able to hoist the `RigidBodySupportState` calls up a level, and this part ran fine on my system, so it may just be a licensing flake as you saw, which could be version dependent; that, or it may be due to the MATLAB bindings not being fully built, thus it tries to fall back on the MATLAB implementation that has a hole in its API.

My suggestion is the following:

- Ensure that you have the MATLAB bindings properly built. On my system, on top of ensuring the prereqs were installed, I had to do the following:
```
$ cd drake-distro
$ mkdir build && cd build
$ export GUROBI_DISTRO=~/Downloads/gurobi6.0.5_linux64.tar.gz
$ cmake .. -G Ninja -DWITH_GUROBI=ON
$ ninja -v  # May fail at director
$ cd drake
$ ninja -v  # Should succeed and install the necessary MATLAB bits

$ cd drake-distro/drake
$ matlab
>> addpath_drake
>> cd examples/Atlas
>> runAtlasWalking
```
- If this still fails, try downgrading to MATLAB R2016b if possible; the documentation at this SHA points to R2015b, so that may be an even more sure-fire way to ensure that it's not due to version differences.

If you can still not get up and running, there may not be much else to be done, as there are no plans to make any modifications to the old MATLAB versions, as indicated above.

Sorry that I can't do more to help!

You may wish to post a request to revive the Atlas implementation in the latest version of Drake here: #6381

closed 2018-12-10 by EricCousineau-TRI

Jerryxiaoyu, 2019-01-02:
@EricCousineau-TRI Sorry for the late reply. 
I recently tried many times as what you suggested and succeeded!
The reason why I failed before might be that I couldn't install prereqs completely. I deleted the original folders and restarted from scratch. 

If anyone wants to reimplement the original matlab-drake distro, here is my suggestion：

-  Download gurobi7.0.2 

- Ensure that the prereqs were installed, and build drake-distro as following:

```
$ cd drake-distro
$ mkdir build && cd build
$ export GUROBI_DISTRO=~/Downloads/gurobi7.0.2_linux64.tar.gz
$ cmake -DWITH_MATLAB=ON -DWITH_MOSEK=ON    -DWITH_AVL=ON -DWITH_YALMIP=ON -DWITH_MESHCONVERTERS=ON -DWITH_XFOIL=ON -DWITH_SIGNALSCOPE=ON -DWITH_VTK=ON -DWITH_GUROBI=ON    -DWITH_SNOPT=OFF -DWITH_IRIS=ON ../.
$ make
```

- Note that `-DWITH_SNOPT` must be set to `OFF` because the snopt repos doesn't exist . The solution is to install SNOPT manually if necessary.

- The URL of IRIS repos should be replaced in  `externals/iris/CMakeList.txt Line28`  using `https://github.com/cddlib/cddlib/archive/0.94h.tar.gz` 

- For MATLAB R2017b(or above),  refer to #5793 #5795 .

Anyway, it works well now. I truly appreciate your help. @EricCousineau-TRI