RobotLocomotion/drake #10167: Documentation needed for using un-normalized quaternion pose_smoother.cc

mitiguy, 2018-12-08:
## Description

In conjunction with landing  PR# 10125
In the file: drake/manipulation/perception/pose_smoother.cc
In the function: Vector7ToIsometry3D
I am unsure why an un-normalized quaternion is being used.
* If the quaternion is not normalized, the associated rotation matrix is not orthogonal.
* If the quaternion is normalized, the following test fails:
drake/manipulation/perception/test/pose_smother_test.cc
* I am unsure if this is a bug or it just needs more documentation..  More notes below.

## Snippet of code.
// Sets a pose from a 7-element array whose first 3 elements are position and
// last 4 elements are a "canonical" quaternion (w, x, y, z).
Isometry3d Vector7ToIsometry3d(const VectorX<double>& pose_vector) {
  Quaterniond quaternion(pose_vector(3), pose_vector(4),
                         pose_vector(5), pose_vector(6));
  Isometry3<double> pose = Isometry3<double>::Identity();
  // TODO(Mitiguy) Ask Naveen to document why the quaternion is not normalized,
  // which allows for quaternion.toRotationMatrix() to return a non-orthonormal
  // matrix.  However, manipulation/perception/test/pose_smoother_test.cc fails
  // (why?) if the next line is uncommented (which makes me question the test).
  // quaternion = quaternion.normalized();
  pose.linear() = quaternion.toRotationMatrix();  // This is not orthonormal!
  pose.translation() = pose_vector.head<3>();
  pose.makeAffine();
  return pose;
}

## More info
The documentation in drake/manipulation/perception/pose_smoother.h
discusses its algorithm as 
"this class implements a simplification based on the version described in
 http://wiki.unity3d.com/index.php/Averaging_Quaternions_and_Vectors
 and in the introduction of Markley et al."

However,  http://wiki.unity3d.com/index.php/Averaging_Quaternions_and_Vectors
says its algorithm "is based on a simplified procedure described in this document:"
https://ntrs.nasa.gov/archive/nasa/casi.ntrs.nasa.gov/20070017872.pdf,
which itself seems to be a simplification.

I am unsure as to how to interpret the results of this test.

If nothing else, it would help to add a note like:
// The algorithm used here depends on the quaternion not being normalized.
// This means that quaternion.ToRotationMatrix() produces a non-orthogonal matrix.
// The theory for not normalizing is (or is not) well understood by the author of this code.
pose.linear() = quaternionToRotationMatrix()

jwnimmer-tri, 2019-03-04:
@naveenoid Please provide guidance here to help us understand what the code is doing / trying to do, per @mitiguy's questions above.  That would at least let another developer help resolve this issue in a timely fashion.

mitiguy, 2019-07-20:
@naveenoid   
I am in the ongoing process of removing Isometry3 from most of the Drake codebase and
instead using RigidTransform which requires an orthogonal rotation matrix.
There is Isometry3 in pose_smoother.cc that is referenced by this issue #10167 
which seems like it cannot be changed.
I am unsure why the test results depend on a non-orthogonal rotation matrix.
You can see related changes in PR# 1186.

jwnimmer-tri, 2021-05-02:
Fixed in #11848.

closed 2021-05-02 by jwnimmer-tri