RobotLocomotion/drake #10177: mbp: Ensure that `GetPositions` / `GetVelocities` returns either a copy or a reference, consistently

EricCousineau-TRI, 2018-12-10:
At present, `MultibodyPlant::GetPositions(...)` and `GetVelocities(...)` have two overloads: `GetX(context) -> VectorBlock<>` (reference) and `GetX(context, model_instance) -> VectorXd` (copy) which have different return types.

For consistency, and to reduce confusion, these should be changed to return the same type, a copy or reference.

Potential solutions:

(1) Hard-to-transition via backwards compatibility: Change them to return copies. Any downstream usages that depend on `GetX(context) -> VectorBlock<>` will silently break; however, it's only been alive for 2 months, so may not be a huge deal.

(2) Make a newly named method, and deprecate (some) of the existing overloads.
(2.a) Deprecate `GetX(context, model_instance)`, make a new method `GetXCopy(context, model_instance)` (and possibly `GetXCopy(context)`).
(2.b) Deprecate `GetX(context)` and `GetX(context, model_instance)`, make a new method `GetXCopy(context)` and `GetXCopy(context, model_instance)`.

My vote is for (2.a).
@edrumwri @sherm1 Can I ask what y'all's votes are?

edrumwri, 2018-12-10:
Eric, it's problematic that they have different return types if this leads to some defect in calling code. Can you describe the calling code problem that you identified f2f?

EricCousineau-TRI, 2018-12-10:
My first-hand experience was making bindings for this; I naively thought they'd have the same semantics; ended up running into https://github.com/pybind/pybind11/issues/1631.

For the return types, if I'm quickly hacking stuff together, I'm going to assume they're a ref or value; me knowing enough about expression templates in Eigen, would know that `auto` should be safe:
```
auto q = plant.GetPositions(context);
auto q_iiwa = plant.GetPositions(context, iiwa);
// Assumed it was a reference, by just looking at `GetPositions` documentation.
plant.GetMutablePositions(context).array() = 0.5;
// `q` is changed but `q_iiwa` is not.
```

In Python, it would look like this:
```
q = plant.GetPositions(context)
q_iiwa = plant.GetPositions(context, iiwa);
plant.GetMutablePositions(context)[:] = 0.5
# `q` is changed, but `q_iiwa` is not.
```

The solution should not be having different semantics in Python vs C++ for the overloads, but rather to fix it at the root-level.

edrumwri, 2018-12-10:
What do you think of having _all_ of the methods for the python code return copies?

jwnimmer-tri, 2018-12-10:
In C++ code at least, if a user calls a `foo = bar.GetStuff(context, ...)` and then mutates the values in `context`, I do not think they have any right to expect that `foo` is unchanged.  There are _hundreds_ of methods like GetStuff that return a reference.  These two under discussion here are not special.

EricCousineau-TRI, 2018-12-10:
> What do you think of having all of the methods for the python code return copies?

Only if the C++ code does the same. I'd prefer not to manage awkward discrepancies if we can avoid them.

> [...] I do not think they have any right to expect that foo is unchanged [...]

I don't understand how this counters the main point of this issue, in that overloads with the same name have the different semantics? My complaint is not about reference semantics, but inconsistency: if they both did the same thing, I wouldn't care.

EricCousineau-TRI, 2018-12-18:
Aye. Haven't found this to be too painful myself since the initial pybind snafu, so I'll go ahead and close this.

closed 2018-12-18 by EricCousineau-TRI