RobotLocomotion/drake #10185: Possible bug with MultibodyPlant

edrumwri, 2018-12-10:
Documentation for CalcPointsGeometricJacobianExpressedInWorld(.) indicates that p_FP_list should be in Frame F.

But Line 685 of multibody_plant.cc reads:

    tree().CalcPointsGeometricJacobianExpressedInWorld(
        context, bodyA.body_frame(), p_WCa, &Jv_WAc);

where 

`p_WCA` is defined on Line 673 above as:

const Vector3<T>& p_WCa = point_pair.p_WCa;

where `point_pair` is of type `PenetrationAsPointPair` and is defined there as:

      /** The point on A that most deeply penetrates B, measured and expressed in
       the world frame. */
      Vector3<T> p_WCa;

The monogram notation indicates that this computation should not be valid.

cc @sherm1 @mitiguy

amcastro-tri, 2018-12-11:
@edrumwri, this method is overloaded for situations like this one in which you have point moving with a given frame (in this case point Ca moves with body frame A) but you already have this point's position expressed in the world frame W. That way you avoid computing point's Ca position in A (p_ACa) which would incur in `CalcPointsGeometricJacobianExpressedInWorld()` computing `p_WCa` again internally.

The overloads are:
- For point position expressed in the frame it moves with [here](https://github.com/RobotLocomotion/drake/blob/master/multibody/tree/multibody_tree.h#L1869) and,
- for the point positions expressed in world frame [here](https://github.com/RobotLocomotion/drake/blob/master/multibody/tree/multibody_tree.h#L1915).

Is this the problem? maybe better docs or API are in order.

edrumwri, 2018-12-11:
Ah... My CLion keeps taking me to the first definition when I click through from MBP! Big fail CLion!

Glad to know this is not a bug!

Three of us (me, Sherm, and Paul) were questioning this code yesterday and the extra output argument didn't jump out at us. Consequently, I'd recommend distinguishing these methods through the API by more than just an output argument.

mitiguy, 2019-06-26:
I think this issue has been addressed with one of the checkboxes in issue #10155, namely:

Merged PR #11623 June 17, 2019 to deprecate two variants of
CalcPointsGeometricJacobianExpressedInWorld(...)
in favor of CalcJacobianTranslationalVelocity( ...).

Merged PR #11671 June 24, 2019 to deprecate 
CalcPointsAnalyticalJacobianExpressedInWorld(...)
in favor of CalcJacobianTranslationalVelocity( ...).

closed 2019-06-26 by mitiguy