RobotLocomotion/drake #10188: Implement kinematic trajectory optimization

tri-ltyyu, 2018-12-11:
Re-implement kinematic trajectory optimization as related to Andres' B-spline approach, or other approaches to interpolate the kinematic trajectory.

hongkai-dai, 2019-03-25:
@avalenzue, do you think we will add B-spline approach to Drake?

jwnimmer-tri, 2019-03-25:
\CC @avalenzu

avalenzu, 2019-03-28:
Yes, my current plan is to work on this at the beginning of April.

tri-ltyyu, 2019-04-29:
Just want to see what the status of this is?

avalenzu, 2019-05-01:
This got pushed back due to other work. I'll get back to it after this week.

jwnimmer-tri, 2020-05-07:
Is this a duplicate of #11827?

avalenzu, 2020-05-07:
No. This issue is coming up with new a new trajectory at planning time, whereas #11827 is for smoothing already feasible trajectories, potentially, during execution.

jwnimmer-tri, 2020-05-07:
Ah, so `class KinematicTrajectoryOptimization` in `planning/kinematic_trajectory_optimization.h`?  I'll add it to our tracking sheet.

avalenzu, 2020-05-07:
Correct.

jwnimmer-tri, 2020-12-09:
Relates to #14431.

PR #18190 (2022-11-01): [planning] Adds KinematicTrajectoryOptimization
Implements kinematic trajectory optimization using
B-spline trajectories.

This is my re-implementation of code written by Andrés Valenzuela
<andres.valenzuela@tri.global>; with a new interface and
implementation, but the same basic formulation.

Resolves #10188.

+@aykut-tri for feature review, please
+@tobiamarcucci for additional math review, please

- - -
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/18190)
closed 2022-11-01 by SeanCurtis-TRI