RobotLocomotion/drake #10189: Geometry Inspector fails to display `sdf` models containing `<collision>` tags

rcory, 2018-12-11:
Running `geometry_inspector` on an `sdf` file that defines `<collision>` tags fails to display the model properly in drake visualizer. If `<collision>` tags are removed, the model displays properly. For instance:

Running the inspector on iiwa14 with no collisions, displays properly:
```bazel run :geometry_inspector -- $HOME/drake/manipulation/models/iiwa_description/sdf/iiwa14_no_collision.sdf```

Running on iiwa14 with collision, fails to display:
```bazel run :geometry_inspector -- $HOME/drake/manipulation/models/iiwa_description/sdf/iiwa14_polytope_collision.sdf```

and produces the following error:
```
warning, cannot find file: ../meshes/collision/link_7_polytope.obj
Traceback (most recent call last):
  File "/home/rcory/.cache/bazel/_bazel_rcory/9d652789e4a2e2b3262bdc71f95b43ca/external/drake_visualizer/lib/python2.7/site-packages/director/lcmUtils.py", line 119, in handleMessage
    callback(msg)
  File "/home/rcory/.cache/bazel/_bazel_rcory/9d652789e4a2e2b3262bdc71f95b43ca/external/drake_visualizer/lib/python2.7/site-packages/director/drakevisualizer.py", line 352, in onViewerLoadRobot
    self.addLinksFromLCM(msg)
  File "/home/rcory/.cache/bazel/_bazel_rcory/9d652789e4a2e2b3262bdc71f95b43ca/external/drake_visualizer/lib/python2.7/site-packages/director/drakevisualizer.py", line 376, in addLinksFromLCM
    self.addLink(Link(link), link.robot_num, link.name)
  File "/home/rcory/.cache/bazel/_bazel_rcory/9d652789e4a2e2b3262bdc71f95b43ca/external/drake_visualizer/lib/python2.7/site-packages/director/drakevisualizer.py", line 299, in __init__
    self.geometry.extend(Geometry.createGeometry(link.name + ' geometry data', g))
  File "/home/rcory/.cache/bazel/_bazel_rcory/9d652789e4a2e2b3262bdc71f95b43ca/external/drake_visualizer/lib/python2.7/site-packages/director/drakevisualizer.py", line 272, in createGeometry
    polyDataList, visInfo = Geometry.createPolyDataFromFiles(geom)
ValueError: need more than 0 values to unpack
```

I've also noticed (when inspecting the DRAKE_VIEWER_LOAD_ROBOT message via lcm-spy) that the `viewer_geometry_data_t` element contains a `string_data` value that indicates a relative mesh file path or full mesh file path, depending on whether the `sdf` file contains collision elements or not, respectively. The visualization fails when the path is relative (and the `sdf` file happens to contain `<collision>` tags).

@RussTedrake @edrumwri @jwnimmer-tri

RussTedrake, 2018-12-11:
can you point us to the sdf?  (or maybe even copy a bit of it here?) 
I'd like to see that the collision meshes are specified in the same way as the visual meshes -- which load correctly?

jwnimmer-tri, 2018-12-11:
FYI The PR #10012 that is underway reworks this tool, but I don't immediately see that it would affect or fix this behavior.

rcory, 2018-12-11:
@RussTedrake You can reproduce the failure with our own iiwa examples (see the issue description above). For example, try `iiwa14_no_collision.sdf` vs `iiwa_polytope_collision.sdf` and you will see the failure.

rcory, 2018-12-11:
A simple `sdf` that reproduces the problem. Removing the `<collision>` element fixes the issue.

```<?xml version="1.0" ?>

<sdf version='1.6'>
  <model name='toy_ex'>
    <link name='toy_link_0'>
      <pose frame=''>0 0 0 0 -0 0</pose>
      <inertial>
        <pose frame=''>0 -0 0.005 0 -0 0</pose>
        <mass>0.04704</mass>
        <inertia>
          <ixx>0.1</ixx>
          <ixy>0</ixy>
          <ixz>0</ixz>
          <iyy>0.1</iyy>
          <iyz>0</iyz>
          <izz>0.1</izz>
        </inertia>
      </inertial>
      <collision name='base_collision'>
        <pose frame=''>0 0 0 0 -0 0</pose>
        <geometry>
          <mesh>
            <scale>1 1 1</scale>
            <uri>../meshes/visual/base.obj</uri>
          </mesh>
        </geometry>
      </collision>
      <visual name='base_visual'>
        <pose frame=''>0 0 0 0 -0 0</pose>
        <geometry>
          <mesh>
            <scale>1 1 1</scale>
            <uri>../meshes/visual/base.obj</uri>
          </mesh>
        </geometry>
      </visual>
    </link>
  </model>
</sdf>
```

soonho-tri, 2018-12-13:
@rcory , FYI, I've assigned you for now.

rcory, 2019-02-05:
Closing this since it seems to have been fixed since this issue was posted.

closed 2019-02-05 by rcory