RobotLocomotion/drake #10202: AddSosConstraint is very slow

hongkai-dai, 2018-12-12:
I have a test that imposes a sum-of-squares (Sos) constraint, and it takes 3 hours to finish one call of `AddSosConstraint`. The majority of time is spent on this line https://github.com/RobotLocomotion/drake/blob/c63627ee938d9273767fede9cba10bddfa8248a0/solvers/mathematical_program.cc#L798
Specifically, when we call `sos_poly == p`, it spends most of time within this loop https://github.com/RobotLocomotion/drake/blob/c63627ee938d9273767fede9cba10bddfa8248a0/common/symbolic_polynomial.cc#L630-L634

I then replaced the line `const auto leq_binding = AddLinearEqualityConstraint(sos_poly == p);` with the following code
```C++
  const symbolic::Polynomial diff{p - sos_poly};
  for (const std::pair<symbolic::Monomial, symbolic::Expression>& item :
       diff.monomial_to_coefficient_map()) {
    AddLinearConstraint(item.second == 0);
  }
```
Namely instead of creating the symbolic formula `sos_poly == p`, I impose the linear constraint directly. And now each call of `AddSosConstraint` takes just a couple of seconds.

To reproduce the slow behavior, you could check out https://github.com/hongkai-dai/drake/commit/91c20bdc16fa42f424485219dc4d9052efbd86e0, and run the bazel target
```
$ bazel build //multibody/rational_forward_kinematics:iiwa_two_boxes_demo
$ ./bazel-bin/multibody/rational_forward_kinematics/iiwa_two_boxes_demo
```
To reproduce the fast behavior, you could check out https://github.com/hongkai-dai/drake/commit/cc0628673efaa1cad06c1876149fa0e11926bfd8, and run the same bazel target.

soonho-tri, 2018-12-17:
https://github.com/RobotLocomotion/drake/blob/c63627ee938d9273767fede9cba10bddfa8248a0/common/symbolic_polynomial.cc#L630-L634

Here is one observation: at line 633, ideally, we want to update `ret` "in-place". However, in the current implementation, it creates a new `Formula` and assign it back to `ret`. Note that each `FormulaAnd` holds an `std::set` object. So this causes `N - 1` unnecessary constructions and destructions of `std::set` in the loop (where `N` is the number of iterations).

Another (more obvious) example is `f = f_1 && f_2 && ... && f_n`. It creates intermediate rvalues such as `(f_1 && f_2 && f_3)` which is safe to be updated in-place if it's not shared, i,e., its `use_count() == 1`.

This is a source of inefficiencies in symbolic classes. For example, `operator+=(e1, e2)` has to check if `e1` is safe to be updated in-place. For now, we don't exploit this yet.

I'm working on an implementation to resolve this issue. I'll open a PR when I think it's ready.

hongkai-dai, 2019-03-08:
@RussTedrake since you are adding SOS examples to underactuated courses, this issue could be relevant.

Inside `AddSosConstraint(p)`, I first create a new SOS polynomial, and then equate the polynomial `p` to this newly created SOS polynomial. Then I return both the Hessian of this new SOS polynomial, together with the linear constraint that equates `p` to the sos polynomial. But to return that linear equality constraint, we must first concatenate the symbolic formulas as in https://github.com/RobotLocomotion/drake/blob/c63627ee938d9273767fede9cba10bddfa8248a0/common/symbolic_polynomial.cc#L630-L634, which can be very slow (as @soonho-tri explained above).

The quick solution is to change the return type of `AddSosConstraint`. Instead of returning both the psd Hessian matrix and the linear equality constraint, if we only return the Hessian matrix, but not the linear equality constraint, I could skip the step to concatenate symbolic formulas, and instead just imposing the linear constraint, but not returning them. This would dramatically improve the speed to add the linear constraint. @RussTedrake do you use the linear equality constraint to equate `p` to the newly created SOS? Is it OK if I remove the linear equality constraint from the return type?

RussTedrake, 2019-03-09:
I do not use the binding that is returned.  It would be fine with me to not return that.

If you are changing the return types, can I make a request to sneak in during the same time?  The PSD matrix returned by `AddSosConstraint(p)` is effectively useless to me, because i don't know the monomial basis that is created behind the scenes.  I would recommend returning the `MonomialBasis` for that call... assuming it does not cause any performance penalty.   I actually almost PR'd this last night!

RussTedrake, 2019-03-09:
fyi @pangtao22

hongkai-dai, 2019-03-09:
> I would recommend returning the MonomialBasis for that call... assuming it does not cause any performance penalty.

Will send the PR to include MonomialBasis

PR #10873 (2019-03-11): change the return type of AddSosConstraint.
Resolves #10202

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/10873)
closed 2019-03-11 by hongkai-dai