RobotLocomotion/drake #10207: python, multibody: Align modules with updated directories

EricCousineau-TRI, 2018-12-12:
With @jwnimmer-tri's improvements to the directory structure, the Python API should match, with backwards compat shims and all that jazz.

jwnimmer-tri, 2018-12-13:
FYI In #9314, I proposed an approach for this where we first _add_ the new spellings, but not port much of anything to use them yet, and nor deprecate the old ones.  That lets us confirm that the old spellings still work, and keeps the PR more focused.  After that, we can perl pie everything in Anzu and Drake to the new spellings, and deprecate the old ones.  Something to consider.

jwnimmer-tri, 2018-12-14:
FYI When fixing this issue, please also update `pydrake_pybind.h` documentation to use a different example for namespace vs package paths for module names.  The multibody_plant example there now is no longer true.

EricCousineau-TRI, 2018-12-16:
Updated with plan of attack. The biggest PRs to review would be to split the `*_py.cc` stuff, and do the `pydrake.attic` moves.

PR #10603 (2019-02-07): py mbp: Deprecate `multibody_tree` modules
Addresses last bullet point, and thus: Resolves #10207.

Requires:
- [x] #10599

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/10603)
closed 2019-02-07 by soonho-tri