RobotLocomotion/drake #10223: `DepthImageToPointCloud` cannot be connected to depth image output ports of `ManipulationStation`

pangtao22, 2018-12-14:
`DepthImageToPointCloud.depth_image_input_port()` can no longer be connected to the `camera_i_depth_image` output port of `ManipulationStation`.

https://travis-ci.org/RobotLocomotion/6-881-examples/jobs/467722327#L6792

pangtao22, 2018-12-14:
@RussTedrake @siyuanfeng-tri

pangtao22, 2018-12-14:
The error message is 
```
RuntimeError: DiagramBuilder::Connect: Mismatched value types while connecting output port camera_1_depth_image of System manipulation_station (type drake::systems::sensors::Image<(drake::systems::sensors::PixelType)5>) to input port depth_image of System drake/perception/DepthImageToPointCloud@00000000075f6060 (type drake::systems::sensors::Image<(drake::systems::sensors::PixelType)6>)
```

jwnimmer-tri, 2018-12-14:
PR #10195 changed the pixel type of that image port from 32F to 16U.  Code that consumes that image will have to adjust accordingly.

pangtao22, 2018-12-14:
It seems that the image type of `DepthImageToPointCloud` is hard coded? It would be nice to add an image type argument to the constructor of `DepthImageToPointCloud`.

jwnimmer-tri, 2018-12-14:
That makes sense.  We would pass in a `sensors::PixelType` enum, and have it handle either both `kDepth16U` and `kDepth32F`.  It will also need a scale factor.

siyuanfeng-tri, 2018-12-14:
That seems like the best option. @pangtao22 are you aware of other users of that output port?

pangtao22, 2018-12-14:
Not really... some students used the depth images for their class projects, but they probably won't touch their code after having submitted their reports?

kmuhlrad, 2018-12-14:
I think the only users went through this system: https://github.com/RobotLocomotion/6-881-examples/blob/master/perception/point_cloud_to_pose_system.py

siyuanfeng-tri, 2018-12-14:
how about we reexpose the 32f port as an orange port (only available in sim)

kmuhlrad, 2018-12-14:
As at least a fix to this problem, I was working on creating a similar system that takes in both a depth image and RGB image and turns them into a PointCloud, now that a PointCloud can store RGB data as well (https://github.com/kmuhlrad/drake/blob/make_rgbd_point_clouds/perception/depth_and_rgb_images_to_point_cloud.cc). In general it probably makes more sense to instead create a Diagram using the existing DepthImageToPointCloud system and combining that with RGB data, but for now I can continue to make a new LeafSystem with the right port types for ManipulationStation.

PR #10250 (2018-12-20): perception, sensors: Improve support for 16U depth images
Closes #10223.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/10250)
closed 2018-12-20 by jwnimmer-tri